gnuradio/gnuradio #1048: CC decoder segfaults unless K=7 and R=2

argilo, 2016-10-07:
The convolutional decoder in gr-fec segfaults if given parameters other than Constraint Length (K) = 7 and Rate Inverse (1/R) = 2. To demonstrate, open gr-fec/examples/fecapi_cc_decoders.grc and change the variable k from 7 to 8. Execute the flowgraph, and it will segfault.

The problem is in cc_decoder_impl.cc, where `d_kernel` takes its value from a map containing only "k=7r=2":

https://github.com/gnuradio/gnuradio/blob/ce354379fee28872ea103eafa9164e6fc1ea54a1/gr-fec/lib/cc_decoder_impl.cc#L150-L158

For other values of K and R, `d_kernel` will end up null. Then the null `d_kernel` is called here, resulting in a segfault:

https://github.com/gnuradio/gnuradio/blob/ce354379fee28872ea103eafa9164e6fc1ea54a1/gr-fec/lib/cc_decoder_impl.cc#L350-L351

jmcorgan, 2016-10-07:
It's clear that non R=1/2, K=7 decoding has not been implemented yet, let alone been VOLK accelerated, so the fix here will be to signal that properly (and document it), instead of the segfault. 

We'd welcome someone picking development up here.

jmcorgan, 2016-10-23:
This was addressed in #1049.

closed 2016-10-23 by jmcorgan