gnuradio/gnuradio #1058: hier_block2 with message port and out put ports

bdyaods, 2016-10-16:
I declared a subclass of hier_block2 with two output ports and on input message port, but when I connected them in a top block, the top block can't start, it said: insufficient connected output ports (2 needed, 0 connected). But I'm sure I have connected the ports correctly.  But it will run succesfully if I do not connect the message port.
   Can you help me? Thanks a lot.

jmcorgan, 2016-10-16:
Please follow the guidelines for reporting issues published at:

http://gnuradio.org/redmine/projects/gnuradio/wiki/ReportingErrors

This may be more suited for the mailing list.

bdyaods, 2016-10-17:
hi, jmcorgan, thank you.
I can't access the website gnuradio.org.
I created a subclass of hier_block2 with python scripts, the blocks can be connected and run successfully, but still could not run in C++.
And I have another question, it is, there is only function message_port_register_hier_in, no set_message_handler, so I can declare a input message port, but how can I process these messages in this block?
Thanks.

jmcorgan, 2016-10-17:
This is very much a question for the GNU Radio mailing list, please take it there.

closed 2016-10-17 by jmcorgan