gnuradio/gnuradio #1059: QtGui: Next Breakage on Qt5 Patch

osh, 2016-10-17:
when building on ubuntu 16.04 x64 with qt4-dev and qt5-dev installed -- qtgui linkage comes out with Qt4 and Qt5 linkage (as shown below)
needless to say this results in instant segfaults upon import
~/next/lib$ ldd libgnuradio-qtgui.so  | grep Qt
	libQt5Widgets.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Widgets.so.5 (0x00007f916c823000)
	libQt5Gui.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Gui.so.5 (0x00007f916a92e000)
	libQt5Core.so.5 => /usr/lib/x86_64-linux-gnu/libQt5Core.so.5 (0x00007f916a458000)
	libQtSvg.so.4 => /usr/lib/x86_64-linux-gnu/libQtSvg.so.4 (0x00007f9169593000)
	libQtOpenGL.so.4 => /usr/lib/x86_64-linux-gnu/libQtOpenGL.so.4 (0x00007f9169291000)
	libQtGui.so.4 => /usr/lib/x86_64-linux-gnu/libQtGui.so.4 (0x00007f916859c000)
	libQtCore.so.4 => /usr/lib/x86_64-linux-gnu/libQtCore.so.4 (0x00007f91680a9000)

osh, 2016-10-17:
building with brand new pybombs prefix and all cmake defaults

jmcorgan, 2016-10-17:
Can you post the CMakeCache.txt from the build directory?

jmcorgan, 2016-10-17:
I've replicated this in the build environment by adding all the qt4 dependencies to the qt5 build tester.

jmcorgan, 2016-11-11:
Fixed in #1092

closed 2016-11-11 by jmcorgan