gnuradio/gnuradio #1076: Message paths remain disconnected after lock/unlock

jaredd, 2016-10-27:
This is an issue created per the [bug described by Eugene Grayver](https://lists.gnu.org/archive/html/discuss-gnuradio/2016-09/msg00136.html) in the discuss-gnuradio mailing list.  A cause and fix was described in [this response from Bastian Bloessl](https://lists.gnu.org/archive/html/discuss-gnuradio/2016-09/msg00207.html) where he suggests copying some code from flat_flowgraph::setup_connections into flat_flowgraph::merge_connections.  I'm too unfamiliar with the merge_connections function to be confident in where to actually insert the loop he suggests.

jmcorgan, 2016-10-27:
Please do not create duplicate issues; I had already addressed this in your previous one #1073.

closed 2016-10-27 by jmcorgan

jaredd, 2016-10-27:
My mistake, I was not paying attention to my issue filter.