gnuradio/gnuradio #1079: top block crash: boost: mutex lock failed in pthread_mutex_lock: Invalid argument

f0urtyfive, 2016-11-02:
I'm able to reliably reproduce a crash I'm seeing in some production code with this test code:
[crash_test_4.py.txt](https://github.com/gnuradio/gnuradio/files/566475/crash_test_4.py.txt)

here is the associated backtraces from gdb:
[gdb bt.txt](https://github.com/gnuradio/gnuradio/files/566476/gdb.bt.txt)

After discussing a bit with tnt in IRC I realized that if I replace main.run() with main.start() and a while True: sleep(1) loop, the crash goes away, so it appears the crash is being caused by some interaction between main.run() and self.unlock() in those different threads.

The test code usually runs anywhere from a dozen to a few hundred iterations before crashing, on my system.

jmcorgan, 2016-11-02:
So, gr.top_block.run() really isn't written to be run the way you are trying to; it is simply a convenience function to replace "start, wait until fg exits or ctrl-c, then stop(), then wait".  In fact the way that you describe changing it to work by @smunaut would be the proper way to do things (except you wouldn't sleep, you'd be doing other things in your program while GR runs in the background.)

That said, this is not really documented, it is at least arguable that what you are trying to is reasonable and run() should be made in such a way to allow this to work.  I'll log this as a bug report but will need to figure out what should be the "proper" behavior and either fix or close as unsupported.

marcusmueller, 2020-10-30:
yeah, in the end that is competing access from python land. Closing as unsupported (for now- if the need really arises, please do open issue)

closed 2020-10-30 by marcusmueller