gnuradio/gnuradio #1080: Support multi-part ZMQ messages

v1bri, 2016-11-04:
It would be nice to add support for multi-part ZeroMQ messages, at least for receiving tag headers in the source blocks.

The current ZeroMQ source and sink blocks are not compatible with multi-part messages. This makes it difficult to write an efficient (ie: zero-copy) data producer that supports stream tags, since the `base_source_impl::load_message` method assumes a tag header will directly precede some data within a single ZeroMQ message. Two memcpy's are required to construct such a message on the producer side (one for tag header and one for data), also requiring 2x the memory.

I'm thinking an updated version of the source blocks would replace `zmq::message_t d_msg` with an `std::vector<zmq::message_t>`, that gets populated in `base_source_impl::load_message` and drained in `base_source::flush_pending`.

If this sounds like a useful feature I'd be glad to throw a pull request together.

jmcorgan, 2016-11-04:
This has been discussed a few times before and in general is a good idea. However, we're going through refactoring of the API parts of some of these blocks in #1038, so anything will have to wait until that gets merged.

I'm inviting @smunaut and @awalls-cx18 to the conversation as they have both done extensive work in this part of the code and have good ideas on where to take this.

v1bri, 2016-11-04:
Ok thanks for the quick reply. I'll keep my changes on hold until #1038 arrives.

jmcorgan, 2017-01-26:
It looks like #1038 is dead in the water, so go ahead and work up something you think would be useful, if you're still interested in this.

commit 6f9303b1 (2017-04-13): Support receiving multi-part ZeroMQ messages

ZeroMQ sink blocks will attempt to load all parts of a multi-part message
before processing tags and outputting items. Allows senders to take
advantage of ZeroMQ's zero-copy message delivery.

Add check for incompatible data sizes between ZMQ endpoints.

Fixes #1080
PR #1211 (2017-04-13): Support receiving multi-part ZeroMQ messages
ZeroMQ sink blocks will attempt to load all parts of a multi-part message
before processing tags and outputting items. Allows senders to take
advantage of ZeroMQ's zero-copy message delivery.

Fixes #1080
closed 2017-04-13 by jmcorgan