gnuradio/gnuradio #1082: Return constellation objects from constellation generators

n-west, 2016-11-04:
The following files have functions that return useful constellations. They should be reworked to return constellation objects and deduped if using the constellation object results in the same constellations.

 * gr-digital/python/digital/psk_constellations.py
 * gr-digital/python/digital/qam_constellations.py

## Rationale

These are useful and common constellations, but the generators return different formats. Having all of them return a constellation object makes them directly usable with the constellation mod/demod blocks.
These are overly specific RRC mod/demods. The canonical way forward is to use the constellation objections with a constellation mod/demod block.

marcusmueller, 2018-08-17:
with a very heavy heart –  because I really think having these two different formats is confusing as hell - postponing to 3.9 :(

gvanhoy, 2019-10-06:
I am not sure how "destructive" I can be here, so I'm just going to show what I think would be easiest to work with as a contributor and/or user. I left the constellation_map_generator in tact, but destroyed all of the function names (psk_2_0x0_0, psk_2_0x0_1, etc..) in favor of arguments.

```python
def psk_constellation(const_points, symbols, k, pi):
    const_points, symbols = constellation_map_generator(const_points, symbols, k, pi)
    return digital.constellation_psk(constellation_points, symbols, len(const_points)).base() 

def const_bpsk(k, pi):
    constellation_points = np.exp(2j * np.pi * np.linspace(0, 1, 2) / 2.0)
    symbols = [0, 1]
    return psk_constellation(constellation_points, symbols, k, pi)

def const_qpsk(k, pi):
    constellation_points = np.exp(2j * np.pi * np.linspace(0, 3, 4) / 4.0)
    symbols = [0, 1, 2, 3]
    return psk_constellation(constellation_points, symbols, k, pi)

def const_8psk(k, pi):
    constellation_points = np.exp(2j * np.pi * np.linspace(0, 7, 8) / 8.0)
    symbols = [0, 1, 3, 2, 6, 7, 5]
    return psk_constellation(constellation_points, symbols, k, pi)
```

marcusmueller, 2023-10-24:
closed as sub-task"y" to #1103

closed 2023-10-24 by marcusmueller