gnuradio/gnuradio #1083: Deprecate and remove MPSK receiver

n-west, 2016-11-04:
Deprecate and remove the C++ MPSK receiver block

* gr-digital/include/gnuradio/digital/mpsk_receiver_cc.h
* gr-digital/lib/mpsk_receiver_cc_impl.cc
* gr-digital/lib/mpsk_receiver_cc_impl.h
* gr-digital/grc/digital_mpsk_receiver_cc.xml

## Rationale

This is a very specific implementation that would be more useful as an example using filter blocks and a constellation mod/demod pair. When deleting ensure that such an example exists.

jmcorgan, 2016-11-12:
Implemented in #1097 for master and #1098 for next.

jmcorgan, 2016-11-15:
Merged.

closed 2016-11-15 by jmcorgan

centromere, 2017-05-08:
> When deleting ensure that such an example exists.

Where may I find this example? @jmcorgan @n-west

intra-au, 2017-11-25:
@n-west @jmcorgan can you please point me to the example using filter blocks and a constellation mod/demod pair (migration path from mpsk_receiver_cc if you will?)

ecarl65, 2018-09-28:
I'm also looking for this example. Was this functionality just removed without the example replacement going in? Despite the original ticket saying not to remove it without preserving the example?

cropinghigh, 2020-11-04:
Sorry, can you please give an example of creating pi/4 DQPSK receiver without mpsk receiver? It it used in osmo-tetra, and i can't find replacement to mpsk_receiver.

Roman-Port, 2021-01-06:
For anybody still stuck on this removal like I was, this block essentially just combined a "Costas Loop" block with a "Clock Recovery MM" block. You can mostly copy the parameters directly from the old "MPSK Receiver" block to these two blocks. Here's an example. The top two blocks produce the same output as the bottom block. For me, I was using this block to implement RDS decoding.

[image]

The ``order`` of the Costas loop will let you pick what kind of PSK signal you're demodulating. I've only tried it with BPSK, order 2, so I might be wrong about this. From the [Costas loop wiki page](https://wiki.gnuradio.org/index.php/Costas_Loop):
* When order=2: used for BPSK where the real part of the output signal is the baseband BPSK signal and the imaginary part is the error signal.
* When order=4: can be used for QPSK where both I and Q (real and imaginary) are outputted.
* When order=8: used for 8PSK.

And finally, here is a full example of demodulating the signal in case it's helpful to anyone. Keep in mind that this example below **has a lot of specific blocks needed just for what I was decoding** that you may not necessarily need. I figure that it might be useful to somebody, though.

[image]

I'm still a little bit frustrated that this block was removed in the first place, though.

cropinghigh, 2021-01-06:
Very much thanks!

cropinghigh, 2021-01-06:
Sadly, this simple solution didn't worked in osmo-tetra. Costas Loop + Clock Recovery MM don't have Theta setting, that is set to pi/4 for pi/4 DQPSK