gnuradio/gnuradio #1086: "done" message syntax changed between master and next

pcercuei, 2016-11-10:
The "device source" block of gr-iio sends itself a "done=false" message, so that it is called again by the scheduler without delay. See https://github.com/analogdevicesinc/gr-iio/blob/threaded/lib/device_source_impl.cc#L303

The problem is, GNU Radio in the master branch expects "done=0" with a PMT of type long, while in the next branch it expects "done=false" with a PMT of type boolean.
Of course, sending the wrong type will trigger an issue and stop the flowgraph.

The handling code of the "done" message should be modified so that it supports both syntaxes, I think.

bastibl, 2016-11-10:
I think that cannot work, since it implies that the next branch should never introduce breaking changes.

The better approach might be to provide a master and a next branch in the OOT module and use them with the corresponding GNU Radio branches.

pcercuei, 2016-11-10:
Well, the next branch should not introduce breaking changes when that can be avoided.

bastibl, 2016-11-10:
There are a lot of breaking changes in the current next branch (a widget framework is gone, submodules are deleted, one scheduler is gone, API changes of the scheduler, etc etc). In my opinion it doesn't make sense to add a lot of code to make everything backward compatible. In fact, it just doesn't work for most of the changes.

At some point, I accepted that an OOT with a single codebase will not work with all versions of GNU Radio. (I tried compile switches based on the GNU Radio versions and other ugly stuff). If you really want your module to be compatible with master and next, I would create branches. You could (in theory) easily work-around your current problem (by patching the module, not GNU Radio), but it might not be the best strategy for software development in the long run.

pcercuei, 2016-11-11:
Yeah, I'm not maintaining two versions. I fixed it with some CMake trickery.

closed 2016-11-11 by pcercuei