gnuradio/gnuradio #1087: Update OOT documentation/tutorial regarding category strings

jmcorgan, 2016-11-10:
See:

http://lists.gnu.org/archive/html/discuss-gnuradio/2016-11/msg00096.html

The use of <category>[HOWTO]</category> in the XML for OOTs is not documented.

kartikp1995, 2016-12-23:
Hi,
I can help with that.

devnulling, 2020-01-19:
@duggabe Would you be able to check on this and/or update the tutorial to included this bit of info?

duggabe, 2020-01-19:
@kartikp1995 If you have access to the #docs channel on Slack or IRC, take a look at the discussion on Jan 15 about XML and YAML in tutorials. then get back with me as 'Barry Duggan' on Slack or 'barry818' on IRC. we appreciate your help!

duggabe, 2020-01-20:
added Note to clarify this issue in https://wiki.gnuradio.org/index.php/OutOfTreeModules#Making_your_blocks_available_in_GRC
This should be ready to close.

777arc, 2020-07-29:
Looks good Barry, thanks!

closed 2020-07-29 by 777arc