gnuradio/gnuradio #1090: Tags for the "next" branch

ilovezfs, 2016-11-11:
We're planning on dropping Qt 4 from [Homebrew](https://github.com/Homebrew/homebrew-core/) soon, so I'm wondering if it would be possible to get some tags on a regular basis for the `next` branch so that we can build gnuradio with Qt 5 without "inventing" releases out of thin air by unilaterally selecting git commit revisions from `next`.

jmcorgan, 2016-11-11:
It's not clear to me what you are asking for--can you elaborate?

ilovezfs, 2016-11-11:
Sure. I don't see any tags listed under https://github.com/gnuradio/gnuradio/releases or https://github.com/gnuradio/gnuradio/tags for the `next` branch. For Homebrew, we always want to build from an upstream tag. Even if the tag is designated devel or beta in this case, it would be much better than our just picking one of your git commits and christening with some release number you didn't come up with and that we invented.

Typically we'll use the GitHub auto-generated source archive tarball that's created whenever upstream makes a new tag. If the software happens to have git submodules that are needed for the build, then we'll build directly from the git tag itself so that the submodules can be cloned.

In other words, I'm hoping you can pick a "good" revision from the next branch and make a tag with a version number for it. It doesn't have to have a prepared distribution tarball, or be designated an official beta release on the gnuradio website, or anything like that.

jmcorgan, 2016-11-11:
Got it.  I need to think a bit on what good tag points are for our next branch, however. The maint/master branches are carefully maintained to ensure compatibility. By design, though, the next branch exists in a quantum state of superposition of working and broken.

ilovezfs, 2016-11-11:
@jmcorgan thanks! Not sure exactly how you handle merging changes from master into next, but those might be good moments to tag.

jmcorgan, 2016-11-11:
We have a continuous merging process.

- Bug fixes PRs are added to the maint branch, which is then merged into master
- PRs for new features in the current generation API are added to master, which is then merged into next
- PRs for new features for the next generation API are added to next

So pretty much every pull request results in a merge to the next branch, or at least groups of them at a time. There really aren't logical, clear-cut tag points until we approach a major release where we'll tag release candidates for testing.

n-west, 2016-11-16:
We could plausibly tag next at the same time as other releases, but they wouldn't necessarily be meaningful rc's.

ilovezfs, 2016-11-17:
That makes sense! Note that https://github.com/Homebrew/homebrew-core/issues/1705 is basically completed and Qt 4 will be removed from Homebrew as soon as https://github.com/Homebrew/homebrew-science/issues/4595 wraps up, if not sooner.

jmcorgan, 2016-11-17:
Just beware there will be some fairly major changes on our next branch in the next few weeks, particularly in terms of dependencies.  We're going through a transition from gtk2 to gtk3 and python2 to python3.

Also, I've added @michaelld to the conversation, as he is our maintainer for Mac related items.

ilovezfs, 2016-11-17:
We have support for gtk3 and python3 so that should be fine. It might be good to get the first `next` tag created before that transition kicks in, though, so we have a baseline.

michaelld, 2016-11-17:
Thanks for the tag, @jmcorgan . I need to update the gnuradio-next port in MacPorts. Been a while since I've worked / played with it.

michaelld, 2016-11-17:
next is Qt5 only now, yes?

jmcorgan, 2016-11-17:
Not quite. Qt5 support is the default, but we need to continue to support Qt4 for our long-term support of Ubuntu 14.04 LTS.  So it would be fine to make it QT5 only for your configurations.

michaelld, 2016-11-17:
I can easily add support for both Qt4 and Qt5, with the former the default for now. Or, just move to Qt5. Qt4 in MacPorts is still maintained and reasonably robust, no matter that it's no longer being actively supported upstream. Qt5 in MacPorts is working, but not as robust yet; it works.

jmcorgan, 2016-11-17:
That's your call--let me know what you decide.

chevdor, 2016-12-05:
Great job @ilovezfs. I tested your new formula (no qt4 installed), it builds and I can start grc.

ilovezfs, 2016-12-05:
@chevdor cool.

Any word here on when there might be a tag?

ilovezfs, 2017-01-27:
Qt 4 was removed from Homebrew on December 30. We of course now have users wondering where gnuradio-companion went. :)

Any update here?

jmcorgan, 2017-01-27:
None yet. We're doing both a maintenance and a feature release shortly; at that time, we can add a tag on next, but it is pretty arbitrary. There's nothing stopping you from picking a commit ID yourself and going with that. As I said, our next branch is a development branch for major feature changes and is not even guaranteed to compile (though it always has.)

ilovezfs, 2017-01-27:
Arbitrary is fine. That would be great.

d13-kevinf, 2017-02-09:
Would you guys be willing to test this before I clean it up and try to make a *proper* pull request out of it? https://github.com/d13-kevinf/homebrew-core/tree/SierraGnuradioQT4Fosphor

ilovezfs, 2017-02-09:
That will not be accepted into Homebrew because it's using Qt 4.

d13-kevinf, 2017-02-09:
The problem with this situation in Homebrew is that as much as I can appreciate your desire to drop QT4 from HomeBrew, IF you choose to *support* gnuradio, wouldn't it be better to either buy into making it work *properly* or just kill it? and remove the formula (for Sierra) until it can be fixed? The existing feel is painful, and quite the regression in functionality. 

MacPorts seems to have a reasonable interim solution as of the past week or so. 
https://trac.macports.org/ticket/53274
https://github.com/macports/macports-ports/blob/master/science/gnuradio/Portfile

There are plenty of 3rd party GNURadio plugins that will not work as a result of this choice as well. This HomeBrew Qt4 choice really neuters the overall functionality of Gnuradio. WxGUI being absent in newer version of GNURadio add in yet an other interesting twist to deal with. Want to use osmocom_fft -F for example? Good luck! 

As an end user, and someone that wasted quite a bit of time to get the formula to work... save some folks the headache and depreciate it for Sierra in the existing state.

ilovezfs, 2017-02-09:
It's really not up for debate at this point. Qt 4 is permanently removed from Homebrew. This is not a proper venue for re-litigating that decision.

>or just kill it?

That is indeed the only alternative to the status quo, other than building from a `next` tag with Qt 5.

As for MacPorts, what they do is entirely up to them, and not relevant to what's going to happen in Homebrew.

d13-kevinf, 2017-02-09:
Oh indeed I wasn't debating your choice to remove QT4... more so suggesting in your transition take into consideration how painful it is for us end users to abruptly loose the ability to have gnuradio-companion simply because you have not worked out a migration path for this specific formula. If a formula doesn't work well because of your transition choices... depreciate it, let us know when you can support it properly. 

What MacPorts IS relevant to us end users as far as the choice to use one package manage vs. the other to get a tool we are accustomed to using. If you can't support us on specific packages, I would argue that you are right I don't care what happens with HomeBrew as I'll likely make use of MacPorts. =] 

Again, not trying to be rude... just suggesting you take into account the *feel* of your transition process to QT5 at the very least for the users of this package. Never mind the other associated headaches. IF things don't work due to dependency issues you won't fix... depreciate them.

d13-kevinf, 2017-02-09:
@ilovezfs "other than building from a next tag with Qt 5" this is unfortunately a crippled choice at best for several plugins.

ilovezfs, 2017-02-09:
I suggest you express such a view on the future PR adding Qt 5 support to the formula.

d13-kevinf, 2017-02-09:
I have the changes queued up... FWIW. Again... QT5 & Gnuradio *break* functionality for a few things that people DO use... me submitting the changes I have for QT5 on next are parallel to how end users have been left out to dry with existing functionality. Is there a better venue to constructively discuss this with you?  

Candidly at this point... the right thing to do seems to take this horse out back for brew and shoot it with regard to the gnuradio.rb Formula specifically. It is unsupportable in its current state (on Sierra at least), and has huge regressions in functionality even if QT5 patches are submitted for future releases.

ilovezfs, 2017-02-09:
Sure. Open a PR removing gnuradio from Homebrew, and make your case.

bastibl, 2017-02-09:
@ilovezfs 
> That is indeed the only alternative to the status quo, other than building from a next tag with Qt 5.

That worked for me until some days ago https://github.com/Homebrew/homebrew-core/commit/d2d15e863109fd2d0fdc24bb754166ba8f4e16e9

Since then, [qwt](https://github.com/Homebrew/homebrew-core/blob/master/Formula/qwt.rb#L17) depends on qt5, while [pyqt5](https://github.com/Homebrew/homebrew-core/blob/master/Formula/pyqt5.rb#L21) still depends on qt5@5.7.

AFAIS, this would link against two qt versions, which, I assume, causes problems.

ilovezfs, 2017-02-09:
The dev versions of sip, qscintilla2, and pyqt5 from riverbank have 5.8 support so that situation should be short lived. If we need to depend on qt@5.7 in qwt for the time being that may be an option. I haven't looked at that specifically.

d13-kevinf, 2017-02-09:
@bastibl likely not useful, but this is how I work around the QWT issue. 
https://github.com/d13-kevinf/homebrew-core/blob/SierraGnuradioQT4Fosphor/Formula/qwt.rb

MikeMcQuaid, 2017-02-09:
> more so suggesting in your transition take into consideration how painful it is for us end users to abruptly loose the ability to have gnuradio-companion simply because you have not worked out a migration path for this specific formula

Don't blame Homebrew for Qt4 being dropped, blame upstream Qt for not continuing to support it on e.g. Sierra without patching and not fixing security issues in QtWebkit. Similarly one could blame GNU Radio for relying on deprecated software. Alternatively (and what I recommend instead): blame no-one as almost everyone working on almost all this software is doing so in their free time and you have the ability to help.

> MacPorts IS relevant to us end users as far as the choice to use one package manager vs. the other to get a tool we are accustomed to using. If you can't support us on specific packages, I would argue that you are right I don't care what happens with HomeBrew as I'll likely make use of MacPorts.

Something worth noting to end-users that's unpleasant to hear but worth stating: if you don't contribute PRs to Homebrew then we don't mind if you move elsewhere as it makes no difference to us. We're not a commercial entity where "more users is better"; we're a community run package manager where your value to that community is directly proportional to the work you contribute to make it better.

michaelld, 2017-02-09:
OK folks: This discussion is no longer relevant to this issue, and, really, the issue is questionably relevant to GNU Radio at all. It's really an issue with Homebrew.

It's not our place to tell HB or MacPorts what to do or how to build / install / use GNU Radio, nor Qt4, nor Qt5 (etc). GR provides a branch with commits that can be used for Qt5 support; not tags, but commits will work. If HB chooses to drop Qt4, or GR, or whatever, that's their issue; not relevant to GR specifically, though I guess it's good for us to know.

Eventually, GR will transition to using Qt5; be patient. Yes, GR uses legacy code; many projects do.

I, as the MP maintainer of GR & a bunch of related ports, will continue to make sure GR functions in that domain as best I can. I won't say MP is perfect -- far from it sometimes; but, it works pretty well overall and I know the platform really well & everything is in 1 place which makes it easier to find and work with. Thus my support for it.

I don't understand the HB model very well, and have general disagreements with some default install parameters. Maybe it's just a lack of understanding / testing on my part, and maybe HB is a great way to go. I don't know, but the discussion here doesn't give me positive vibes in that regard.

@jmcorgan : My advice is to close this issue as "wontfix" or whatever GitHub does. I think it's time to be done here.

ilovezfs, 2017-02-09:
Sorry, but above it was stated "None yet. We're doing both a maintenance and a feature release shortly; at that time, we can add a tag on next, but it is pretty arbitrary."

If that's not of interest to MacPorts, so be it, but it still is to Homebrew. Thanks.

d13-kevinf, 2017-02-09:
@MikeMcQuaid  the irony of "Something worth noting to end-users that's unpleasant to hear but worth stating: if you don't contribute PRs to Homebrew then we don't mind if you move elsewhere as it makes no difference to us" is that I entered this conversation with a possible PR request that I was asking folks to review. People in the community have already thanked me for a working solution... apparently that is neither here nor there with regard to Brew standards. 

@michaelld "the issue is questionably relevant to GNU Radio at all. It's really an issue with Homebrew....not relevant to GR specifically, though I guess it's good for us to know." truth... thanks for the sanity check. The reality is "Eventually, GR will transition to using Qt5; be patient. Yes, GR uses legacy code; many projects do." and until that happens... Brew is unable to adequately support us as end users with the gnuradio.rb formula specifically mentioned. 

@ilovezfs I am finding a bit more irony in you being the end user as it were to the gnuradio repo FWIW. "If that's not of interest to MacPorts, so be it, but it still is to Homebrew". It feels sort of like how it is of interest to us brew end users that we are basically being told to suck it up and deal with lack of gnuradio functionality in a "won't fix" fashion just the same. 

Am I obtuse to think that it would be trivial of you all (Brew maintainers) to add a "legacy" tap section in which the community supported QT4 patches / packages *could* be used by folks that so chose to do so? If there is a better "issue" on GitHub to discuss these semantics please point me to it.

d13-kevinf, 2017-02-09:
Likewide @MikeMcQuaid the one Pull Request for this is marked as "Do Not Merge" and even though my work fixes some of his nuances it is clear it was instantly NOT considered due to semantics... so to throw that in my face re Pull Requests is kinda silly... 
https://github.com/Homebrew/homebrew-core/pull/7976

@ilovezfs Followed up claiming the issue was in essence solved and should not hold up further progress: "I'd prefer we not do this. gnuradio/gnuradio#1090 is already in the works, and I've already re-written this formula for Qt 5 compatibility locally pending resolution of that issue."

Which clearly isn't the case or we wound not be having this discussion in the first place. 
https://github.com/Homebrew/homebrew-core/pull/7976#issuecomment-267883091

MrSpock, 2017-02-12:
Is there any clear statement about what will be done with qt4/gnuradio. gnuradio GUI is one ow most important piece of software for anyone who is doing anything related to RF/DSP.

Just send us clear message should we ditch homebrew because of this or there is reasonable time frame to fix this.

d13-kevinf, 2017-02-13:
@MrSpock I think that the statements from @ilovezfs are pretty clear... in his opine he has already re-written the formula and to him this is a "resolution" path. I can assume he does not actually use GNURadio (extensively with a number of 3rd party .GRC flow graphs, etc) so full groking why lack of QT4is an issue (and why QT5 is not a resolution path) seems to miss the mark for him and other non gnuradio users involved in brew project maintenance. 

Read the statements though, the stance is clear: 
"I'd prefer we not do this. gnuradio/gnuradio#1090 is already in the works, and I've already re-written this formula for Qt 5 compatibility locally pending resolution of that issue." - https://github.com/Homebrew/homebrew-core/pull/7976#issuecomment-267883091

"Qt 4 will be removed from Homebrew as soon as Homebrew/homebrew-science#4595 wraps up, if not sooner." - https://github.com/gnuradio/gnuradio/issues/1090#issuecomment-261113525

If you read: https://github.com/Homebrew/homebrew-science/issues/4595 you can see they wanted to fast track QT4 out of the project as quickly as possible... 

@MikeMcQuaid mentions "This looks like the last one with a hard dependency. It'll probably need boneyarded unfortunately if it gets to e.g. Christmas and there's no workaround here." regarding qgis - https://github.com/Homebrew/homebrew-science/issues/4595#issuecomment-265265397

I think "sounds like it's time to move on then." frames the mentality quite well? 

But lets be clear... as of December 6th, this was all a done deal. QT4 is dead for HomeBrew, and Brew's stance re: legacy support is overly obvious. "Transition to Qt 5 #4595" https://github.com/Homebrew/homebrew-science/issues/4595 was the only thing holding up "Transition formulae to Qt 5 #1705" - https://github.com/Homebrew/homebrew-core/issues/1705

When this issue was closed... THIS gnuradio.rb thing became a nuisance for the project maintainers. They want it to be gone, and seem to be unable to empathize with the folks that were using it in the previous form. https://github.com/Homebrew/homebrew-science/issues/4595#event-890031404

They pretty well gave us the "How Open Source Projects Survive Poisonous People" standard methodology in a nutshell. They have things to do, that don't involve our concerns as noisy end users! 

https://www.youtube.com/watch?v=-F-3E8pyjFo

If you are in a pinch... it seems we are going to have to maintain our own forks to keep a *working* version of Gnuradio-Companion and a healthy set of the compiled modules intact. 
https://github.com/d13-kevinf/homebrew-core/tree/SierraGnuradioQT4Fosphor

MrSpock, 2017-02-13:
@d13-kevinf: Thanks for clarification. Most important info is about your repository. I've spent better part of yesterday evening looking for some repo that will solve my problem. Tried hack-rf & other. I will try yours :)

d13-kevinf, 2017-02-13:
If you have problems with it, let me know... I want this to be something that *works* for people that need it. Shoot me an issue in the repo if you have a specific conflict, or problem.  https://github.com/d13-kevinf/homebrew-core/issues

jmcorgan, 2017-02-13:
Gentlemen, just to clarify before closing this.

GNU Radio 3.8 release, currently being developed on the 'next' branch, will support QT5.  It is possible to build this and work with QT5 today.  However, our next branch is considered unstable, we will be making some significant changes to it soon for Python3 and gtk3 support, which will break many things that expect the stability of the typical releases we cut from our 'master' branch.

In other words, you are of course free to distribute whatever you wish, but until our next branch is stabilized with all the dependency changes planned for 3.8, it *will* break for you in unexpected ways. There are several major branch merges coming for things like Python3 and gtk3. We will not add a tag to this branch until such a time as we feel it can be relied upon for users to develop GNU Radio applications against.

I don't think there is much further to discuss on this, so I am closing this.

closed 2017-02-13 by jmcorgan

michaelld, 2019-10-17:
> As gnuradio 3.8 has been released, should it now work on homebrew?

Short answer: Yes!

Hopefully someone has figured (is figuring?) out how to do that!

Clovel, 2020-05-06:
Still doesn't work as of 07/05/2020. Homebrew still installs only `GNURadio-config-info`.

OS: macOS Catalina 10.15.4 (19E287).