gnuradio/gnuradio #1094: Segfault in gnuradio-runtime (block.cc)

mrjacobagilbert, 2016-11-11:
I have a GR application that is throwing a segmentation fault while otherwise running correctly. It runs for somewhere between 10 seconds and ~5 minutes but consistently ends with the following SIGSEGV:

```
Program received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7fddf1ffb700 (LWP 1662)]
shared_count (r=..., this=<optimized out>) at /opt/gnuradio/include/boost/smart_ptr/detail/shared_count.hpp:382
382	        if( pi_ != 0 ) pi_->add_ref_copy();
```

The full GDB backtrace is:

```
(gdb) bt
#0  shared_count (r=..., this=<optimized out>) at /opt/gnuradio/include/boost/smart_ptr/detail/shared_count.hpp:382
#1  shared_ptr (this=<optimized out>) at /opt/gnuradio/include/boost/smart_ptr/shared_ptr.hpp:328
#2  detail (this=0x32a3d78) at /opt/gnuradio/src/gnuradio/gnuradio-runtime/include/gnuradio/block.h:839
#3  gr::block::finished (this=0x32a3d78) at /opt/gnuradio/src/gnuradio/gnuradio-runtime/lib/block.cc:753
#4  0x00007fde8a1cb21c in gr::tpb_thread_body::tpb_thread_body (this=0x7fddf1ffadb0, block=..., 
    max_noutput_items=<optimized out>) at /opt/gnuradio/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:129
#5  0x00007fde8a1bd511 in operator() (this=0x3cdd890)
    at /opt/gnuradio/src/gnuradio/gnuradio-runtime/lib/scheduler_tpb.cc:44
#6  operator() (this=0x3cdd890)
    at /opt/gnuradio/src/gnuradio/gnuradio-runtime/include/gnuradio/thread/thread_body_wrapper.h:51
#7  boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke
    (function_obj_ptr=...) at /opt/gnuradio/include/boost/function/function_template.hpp:153
#8  0x00007fde8a170e80 in operator() (this=<optimized out>)
    at /opt/gnuradio/include/boost/function/function_template.hpp:767
#9  boost::detail::thread_data<boost::function0<void> >::run (this=<optimized out>)
    at /opt/gnuradio/include/boost/thread/detail/thread.hpp:117
#10 0x00007fde894b60da in thread_proxy () from /opt/gnuradio/lib/libboost_thread-mt.so.1.53.0
#11 0x00007fde9ed96182 in start_thread (arg=0x7fddf1ffb700) at pthread_create.c:312
#12 0x00007fde9eac347d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:111
```

jmcorgan, 2016-11-12:
Stack frames 4-12 are normal. Frame 3 happens during flowgraph shutdown.  Are you doing any sort of flowgraph reconfiguration?

mrjacobagilbert, 2016-11-12:
Not intentionally. As far as we can tell everything is running as normal until this happens.

mrjacobagilbert, 2016-11-14:
Upon further investigation, it appears that [this function](https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/block.cc#L751) is not shutdown related, rather end-of-work-call related. I added a print statement in there and it is called [a lot] during FG execution.

bastibl, 2016-11-14:
Note that the function is called finished() like is_finished(), not finish() like shutdown the flowgraph.
It's called regularly when the scheduler checks if a block was shutdown trough the system port.  So this sounds OK to me.

jmcorgan, 2016-11-14:
I think the last suggestion was to compile this against the next branch to see if any of the runtime changes there affect the problem.

mrjacobagilbert, 2016-11-15:
I attempted to rebuild using next, however I have a different segfault related to the QT GUI:

```
>>> from gnuradio import qtgui

Program received signal SIGSEGV, Segmentation fault.
0x00007ffff0d295a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
(gdb) 
(gdb) bt
#0  0x00007ffff0d295a4 in ?? () from /usr/lib/x86_64-linux-gnu/libQtGui.so.4
#1  0x00007ffff7de74ea in call_init (l=<optimized out>, argc=argc@entry=1, 
    argv=argv@entry=0x7fffffffdc48, env=env@entry=0x7fffffffdc58)
    at dl-init.c:72
#2  0x00007ffff7de75fb in call_init (env=0x7fffffffdc58, argv=0x7fffffffdc48, 
    argc=1, l=<optimized out>) at dl-init.c:30
#3  _dl_init (main_map=main_map@entry=0xa69750, argc=1, argv=0x7fffffffdc48, 
    env=0x7fffffffdc58) at dl-init.c:120
#4  0x00007ffff7dec712 in dl_open_worker (a=a@entry=0x7fffffffc600)
    at dl-open.c:575
#5  0x00007ffff7de7394 in _dl_catch_error (
    objname=objname@entry=0x7fffffffc5f0, 
    errstring=errstring@entry=0x7fffffffc5f8, 
    mallocedp=mallocedp@entry=0x7fffffffc5ef, 
    operate=operate@entry=0x7ffff7dec300 <dl_open_worker>, 
    args=args@entry=0x7fffffffc600) at dl-error.c:187
#6  0x00007ffff7debbd9 in _dl_open (
    file=0x7ffff644e134 "/opt/gr-next/lib/python2.7/dist-packages/gnuradio/qtgui/_qtgui_swig.so", mode=-2147483646, 
    caller_dlopen=0x522aa3 <_PyImport_GetDynLoadFunc+243>, nsid=-2, 
    argc=<optimized out>, argv=<optimized out>, env=0x7fffffffdc58)
    at dl-open.c:660
#7  0x00007ffff75edf09 in dlopen_doit (a=a@entry=0x7fffffffc830) at dlopen.c:66
---Type <return> to continue, or q <return> to quit---
#8  0x00007ffff7de7394 in _dl_catch_error (objname=0x9f1240, errstring=0x9f1248, mallocedp=0x9f1238, 
    operate=0x7ffff75edeb0 <dlopen_doit>, args=0x7fffffffc830) at dl-error.c:187
#9  0x00007ffff75ee571 in _dlerror_run (operate=operate@entry=0x7ffff75edeb0 <dlopen_doit>, args=args@entry=0x7fffffffc830)
    at dlerror.c:163
#10 0x00007ffff75edfa1 in __dlopen (file=<optimized out>, mode=<optimized out>) at dlopen.c:87
#11 0x0000000000522aa3 in _PyImport_GetDynLoadFunc ()
#12 0x000000000052264f in _PyImport_LoadDynamicModule ()
#13 0x00000000005c6815 in ?? ()
#14 0x00000000004c41da in PyEval_EvalFrameEx ()
#15 0x00000000004c994f in PyEval_EvalFrameEx ()
#16 0x00000000004c22e5 in PyEval_EvalCodeEx ()
#17 0x00000000004c2089 in PyEval_EvalCode ()
#18 0x00000000004c019b in PyImport_ExecCodeModuleEx ()
#19 0x00000000004bd24e in ?? ()
#20 0x00000000004af395 in PyImport_ImportModuleLevel ()
#21 0x00000000004b10a8 in ?? ()
#22 0x00000000004b0de3 in PyObject_Call ()
#23 0x00000000004ce140 in PyEval_CallObjectWithKeywords ()
#24 0x00000000004c6d01 in PyEval_EvalFrameEx ()
#25 0x00000000004c22e5 in PyEval_EvalCodeEx ()
#26 0x00000000004c2089 in PyEval_EvalCode ()
#27 0x00000000004c019b in PyImport_ExecCodeModuleEx ()
#28 0x00000000004bd24e in ?? ()
#29 0x00000000004be547 in ?? ()
#30 0x00000000004afd2d in ?? ()
#31 0x00000000004af969 in PyImport_ImportModuleLevel ()
#32 0x00000000004b10a8 in ?? ()
#33 0x00000000004b0de3 in PyObject_Call ()
#34 0x00000000004ce140 in PyEval_CallObjectWithKeywords ()
#35 0x00000000004c6d01 in PyEval_EvalFrameEx ()
#36 0x00000000004c22e5 in PyEval_EvalCodeEx ()
#37 0x00000000004c2089 in PyEval_EvalCode ()
#38 0x00000000004f1e6f in ?? ()
#39 0x000000000044c7b0 in PyRun_InteractiveOneFlags ()
#40 0x000000000044c575 in PyRun_InteractiveLoopFlags ()
#41 0x000000000042e91d in ?? ()
#42 0x000000000049e36a in Py_Main ()
#43 0x00007ffff7811830 in __libc_start_main (main=0x49dcd0 <main>, argc=1, argv=0x7fffffffdc48, init=<optimized out>, 
    fini=<optimized out>, rtld_fini=<optimized out>, stack_end=0x7fffffffdc38) at ../csu/libc-start.c:291
#44 0x000000000049dbf9 in _start ()
```

This is via pybombs, and since 3.7 builds I apt-get installed pyqt5-dev and python-pyqt5 (qtgui did not build without these).

jmcorgan, 2016-11-15:
You probably did this, but with a major version change like this you need to make sure that all the old binaries are removed, *and* do the build completely from scratch (blow away the build dir). 

Finally, the packages for qt5 support are:

libqwt-qt5-dev
qtbase5-dev
python-pyqt5
pyqt5-dev-tools

It's ok if you leave the qt4 dependencies installed, but only if you use the very latest next branch as of a day or so ago.

jmcorgan, 2016-11-15:
Alternatively, you can add -DDESIRED_QT_VERSION=4 to your cmake command line to compile against QT4.

Pybombs does not support next branch yet, so you'll need to do all of the above manually.

mrjacobagilbert, 2016-11-15:
OK, so I built next successfully using the QT4 as described above (qtgui segfaulted for me upon import with qt5...separate issue, probably on my end) and I get a whole variety of new segfaults. I ran this 4 times and got 4 different issues, here is one that did fail in a similar though not identical way:

```
Thread 59 "usrp_source2" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f5778ff9700 (LWP 27360)]
boost::detail::shared_count::shared_count (r=..., this=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:454
454	        if( pi_ != 0 ) pi_->add_ref_copy();
(gdb) bt
#0  boost::detail::shared_count::shared_count (r=..., this=<optimized out>)
    at /usr/include/boost/smart_ptr/detail/shared_count.hpp:454
#1  boost::shared_ptr<gr::block_detail>::shared_ptr (this=<optimized out>)
    at /usr/include/boost/smart_ptr/shared_ptr.hpp:323
#2  gr::block::detail (this=<optimized out>)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/include/gnuradio/block.h:863
#3  gr::tpb_detail::notify_downstream (this=this@entry=0x44932e0, d=d@entry=0x44932d0)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/tpb_detail.cc:60
#4  0x00007f5824c45eb1 in gr::tpb_detail::notify_neighbors (this=this@entry=0x44932e0, d=d@entry=0x44932d0)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/tpb_detail.cc:67
#5  0x00007f5824c48785 in gr::tpb_thread_body::tpb_thread_body (this=0x7f5778ff8dc0, block=..., 
    max_noutput_items=<optimized out>)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:138
#6  0x00007f5824c38545 in gr::tpb_container::operator() (this=0x4728810)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/scheduler_tpb.cc:44
#7  gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (this=0x4728810)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/include/gnuradio/thread/thread_body_wrapper.h:51
#8  boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke (function_obj_ptr=...) at /usr/include/boost/function/function_template.hpp:159
#9  0x00007f5824bd6970 in boost::function0<void>::operator() (this=<optimized out>)
    at /usr/include/boost/function/function_template.hpp:773
#10 boost::detail::thread_data<boost::function0<void> >::run (this=<optimized out>)
    at /usr/include/boost/thread/detail/thread.hpp:116
#11 0x00007f582415c5d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#12 0x00007f583a9fd70a in start_thread (arg=0x7f5778ff9700) at pthread_create.c:333
#13 0x00007f583a73382d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

mrjacobagilbert, 2016-11-15:
This backtrace may also be related:

```
Thread 31 "extract_metada4" received signal SIGSEGV, Segmentation fault.
[Switching to Thread 0x7f0167fff700 (LWP 27221)]
gr::tpb_thread_body::tpb_thread_body (this=0x7f0167ffedc0, block=..., max_noutput_items=<optimized out>)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:100
100	        if(block->has_msg_handler(i.first)) {
(gdb) bt
#0  gr::tpb_thread_body::tpb_thread_body (this=0x7f0167ffedc0, block=..., max_noutput_items=<optimized out>)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/tpb_thread_body.cc:100
#1  0x00007f01afe7d545 in gr::tpb_container::opeator() (this=0x3e35fa0)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/lib/scheduler_tpb.cc:44
#2  gr::thread::thread_body_wrapper<gr::tpb_container>::operator() (this=0x3e35fa0)
    at /opt/gr-next/src/gnuradio/gnuradio-runtime/include/gnuradio/thread/thread_body_wrapper.h:51
#3  boost::detail::function::void_function_obj_invoker0<gr::thread::thread_body_wrapper<gr::tpb_container>, void>::invoke (function_obj_ptr=...) at /usr/include/boost/function/function_template.hpp:159
#4  0x00007f01afe1b970 in boost::function0<void>::operator() (this=<optimized out>)
    at /usr/include/boost/function/function_template.hpp:773
#5  boost::detail::thread_data<boost::function0<void> >::run (this=<optimized out>)
    at /usr/include/boost/thread/detail/thread.hpp:116
#6  0x00007f01af3a15d5 in ?? () from /usr/lib/x86_64-linux-gnu/libboost_thread.so.1.58.0
#7  0x00007f01c5c4270a in start_thread (arg=0x7f0167fff700) at pthread_create.c:333
#8  0x00007f01c597882d in clone () at ../sysdeps/unix/sysv/linux/x86_64/clone.S:109
```

mrjacobagilbert, 2016-11-17:
After further testing, we started getting segfaults from SSE2 and kernel routines and moved development to a different hardware platform. The listed backtraces do not appear on this new system so we now consider this issue done with. I am going to leave the issue open until I can re-test with gr-next (jcorgan request), and I will post results and close at that point.

jmcorgan, 2016-11-20:
I'll go ahead and close this; if you get tests done on next branch you'll still be able to add the results here.

closed 2016-11-20 by jmcorgan