gnuradio/gnuradio #1102: Deprecate and remove cpfsk_bc; replace with hier block using frequency_mod_fc

n-west, 2016-11-16:
The cpfsk_bc block is an implementation of 2cpfsk. It is currently marked as deprecated as part of the ongoing 3.8 cleanup. It should be removed on next, but only with a suitable replacement for "out of the box" cpfsk. The replacement will likely look something like the following:

(This is only a suggestion and actual implementation should be designed by assignee)
hier block with byte input -> repeat (samples/per symbol) -> chunks to symbols -> frequency mod -> complex output.

The parameters should be preserved as they match theory and common descriptions of the modulation (modulation index, samples per symbol). As a starting point, the modulation index to sensitivity of the frequency modulators is sensitivity = k * pi / samples_per_symbol. As an added feature the new hier block should be able to generate cpfsk with a higher order than 2 which can be accomplished with the chunks to symbols symbol table. In this case it makes sense for a symbol table to be a parameter.

jmcorgan, 2016-11-16:
This is a great write up.  

I'd clarify that the hierblock 'byte' input is actually a channel symbol input (takes on discrete values of 0, 1, 2, ... depending on the order of the FSK), which then gets mapped by a configurable symbol table to modulation symbols (real valued frequency values) that then get passed to the FM modulator for complex output.

We've had a longstanding terminology issue of calling discrete channel symbols "chunks".  I'd like to see us consistent in the 3.8 release of calling these channel symbols, that then get mapped to modulation symbols. In the case of PSK these are actually constellation points; in the case of FSK here they are real valued baseband frequencies.  Nonetheless, channel symbol=>modulation symbol is general enough to capture any kind of modulation.

jmcorgan, 2016-11-16:
I'll also add that since the current cpfsk_bc block is C++, the replacement hierblock should be C++.

n-west, 2016-11-16:
For reference, an alternative implementation is to use cpm_mod in cpm.py with a thin hier-block wrapper that sets cpm_type = 0. If that is used a conversion of cpm.py to a C++ hierblock would be required.

n-west, 2017-03-15:
https://github.com/n-west/gnuradio/commit/b6b6c895e2b811085fa2225d3f495b04bead5bf0 will be tested on a couple of machines in addition to my ancient devvy. Hopefully I can resurrect an e310 to test on ARM. This is will lead to plowing away this, #1103 and #1104 as well

jmcorgan, 2017-03-17:
Great, glad to see progress in this area.

marcusmueller, 2018-08-17:
@n-west this becomes very dear to my current getting-things-done efforts :)

So, that frequency_mod work was great, how do we progress with this?

marcusmueller, 2023-10-24:
closed as sub-task"y" to #1103

closed 2023-10-24 by marcusmueller