gnuradio/gnuradio #1106: Deprecate packet synchronizers

n-west, 2016-11-16:
During gr-digital 3.8 cleanup talks the following blocks were ruled to be removable:

* simple_correlator
* simple_framer
* mod/demod_pkts (in python/digital/pkt.py)
* correlate_and_sync (replaced by corr_est)
* correlate_access_code_bb
* correlate_access_code_bb_ts
* correlate_access_code_ff_ts

Most of these are either an old way of dealing with packets, very specific to something no one cares about, or replaceable with the new packet parsing tools. When deprecating and removing ensure that any examples are replaced with something that is *not* deprecated.

awalls-cx18, 2016-12-18:
Just some user input:
A correlate_access_code_* block is usually employed by me post clock recovery and slicer, as tags added by corr_est prior to timing recovery usually get pushed around due to imperfect tag propagation through the clock recovery blocks.

I haven't looked hard at what the replacement might be,

n-west, 2016-12-19:
Correlate_access_code_bb_tag will remain, which is the only one worth keeping in-tree.

 * bb_ts is actually looking for a specific frame structure.
 * ff_ts is also looking for that frame structure.
 * _bb is the only block that operates the way it does-- just use the tag version.

nowls, 2017-01-26:
I'll add that we should have a corresponding _ff version in case you don't want to slice beforehand (to have soft input to a FEC block, say). correlate_access_code_bb also looks like it needs to be volkified like the other correlate_access_code_* blocks.

jmcorgan, 2017-01-26:
Hey, Sean, great to hear from you. Would you be interested in helping out with some of this?

nowls, 2017-01-26:
Yes, I can help! I agree with @n-west that _tag_bb is the only one worth keeping around. Giving it a _tag_ff version should be straightforward. Which release should I target that for? I don't think it needs to wait for 3.8.

jmcorgan, 2017-01-26:
A PR for a new block should be against master and will get into 3.7.12.

marcusmueller, 2018-08-17:
@nowls um, what's the state of this? I'm a bit lost on wether we already have such a `_tag_ff` implementation or not.

marcusmueller, 2018-08-18:
cc @n-west as I have no idea whether I should be doing something here or not, but you might know.

nowls, 2018-08-29:
I added `correlate_access_code_tag_ff` in #1189 but didn't do any work deleting stuff for 3.8.

Dantali0n, 2021-12-15:
How do we now perform clock recovery? as both clock recovery modules are in the deprecated category yet some tutorials still use these?

marcusmueller, 2023-10-24:
kind of like #1103 as in, needs time to check up on state and discussion with @mormj and @willcode

mbr0wn, 2024-03-25:
@marcusmueller I think we can close this. We missed the opp to remove this when we released 3.10, and removing them now would break the API. For GR 4, we would obviously not port them.

marcusmueller, 2024-03-25:
@mbr0wn fair enough an assessment. I'll quickly make a new issue, "make sure all these are marked deprecated" and close this one.

marcusmueller, 2024-03-25:
oh wait too fast this is a deprecation issue. No, let's not close it then, lets just make a PR that deprecates them. Deprecation hurts nobody seriously, this has been consensus forever, and at least we don't lead people onto deadend approaches.

marcusmueller, 2024-03-25:
For beginners coming this way: "deprecation" just means that for each single block mentioned

1. we add the `flags: [deprecated]` property to each block's gr-something/grc/something_blockname.block.yml,
2. search the examples whether any existing example uses the block
3. open a new issue "remove or update example XYZ to not use deprecated block"

while doing this, talking to the people in the #docs Matrix chat room is probably a good idea, they have a good idea of existing examples.

Gowri213, 2025-12-23:
Hi  @marcusmueller,  I’m interested in working on this issue.
Could you please confirm if it’s still open and unassigned?
I’d be happy to take it up. Thank you!

Gowri213, 2025-12-24:
Ok, Thank you!

ffu, 2025-12-27:
For your consideration: the correlate access code - tag stream block is still relevant because it can deframe a packet that uses the default header format. For many modulation types (psk, gmsk, gfsk), it's the easiest way to implement packet reception. For these reasons, this block is used in the packet communications tutorial and other similar tutorials.