gnuradio/gnuradio #1110: ControlPort: setup_rpc() is not called automatically for blocks without stream connections

mrjacobagilbert, 2016-11-21:
The loop for which setup_rpc() is called on blocks is dependent on the block being in the flowgraph edgelist (see: https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L593). This means setup_rpc() will not be called for blocks that only use message ports for communication, or are otherwise not appear in the flowgraph edges() object.

jmcorgan, 2016-11-21:
It looks like you just need to concat the edges and msg_edges into a single vector to run across in the loop to call setup_rpc().  Want to take a stab at this?

jmcorgan, 2016-11-21:
Concat and unique sort, that is.

bastibl, 2016-11-21:
a bit further down in the function, it looks like the code already iterates over such a list:
https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L796

maybe RPC initialization could be moved there.

mrjacobagilbert, 2016-11-22:
I'll submit a PR for a fix, but its probably not the fix that is desirable long term. I just added the same code that calls setup_rpc() if not already when the edges are iterated over here: https://github.com/gnuradio/gnuradio/blob/master/gnuradio-runtime/lib/hier_block2_detail.cc#L746.

It seems like a better long term solution would be to do as @bastibl suggested and loop over all the blocks once, but this is probably best left to the more software inclined (I tried: segfault, revert).

jmcorgan, 2017-01-12:
Fixed in #1119

closed 2017-01-12 by jmcorgan