gnuradio/gnuradio #1111: Pack k bits tag propagation error

jaredd, 2016-11-21:
Pack k bits uses the default mechanism for propagating tags which uses the relative rate to determine the output index based on the input index.  It seems, however, that the output index is determined as the round(input_index * relative rate) which means if I have an input with a tag on 5 through a pack 2 bits block, my tag comes out on index round(5 * 1/2) = 3 rather than index 2 (where it's the 1st bit of the output item).  Really, the output tags should be associated with the word containing the bit the tag was on at the input.

noc0lour, 2016-12-17:
Basically this can be solved by using a floor division instead of round()?

jaredd, 2016-12-31:
In this case yes, but the tag propagation will have to be moved into the block unless the default behavior for all tag propagation is to be changed, which I don't think would be correct for other blocks.

jasonuher, 2021-02-24:
We also ran into this issue.  I'll see if I can get the fix (may have just eliminated the kernel all together so who knows if that's useful here).

I can definitely share my test that checks for the problem though:

```    def test_004(self):
        # Test tags propagation
                                             #(offset, key,value,src)
        src_data = [1, 0, 1, 1, 0, 0, 0, 1]
        test_tags = list()
        test_tags.append(
                gr.tag_utils.python_to_tag((1, pmt.intern('tag_byte'), pmt.from_long(0), None)))
        test_tags.append(
                gr.tag_utils.python_to_tag((2, pmt.intern('tag_byte'), pmt.from_long(1), None)))
        test_tags.append( 
                gr.tag_utils.python_to_tag((3, pmt.intern('tag_byte'), pmt.from_long(1), None)))
        test_tags.append(
                gr.tag_utils.python_to_tag((5, pmt.intern('tag_byte'), pmt.from_long(2), None)))
        test_tags.append(
                gr.tag_utils.python_to_tag((6, pmt.intern('tag_byte'), pmt.from_long(3), None)))
        expected_tag_offsets = [0, 1, 1, 2, 3]
    
        expected_results = [2, 3, 0, 1]
        src = blocks.vector_source_b(src_data, False,1, test_tags )
        op = blocks.pack_k_bits_bb(2)
        dst = blocks.vector_sink_b()
        self.tb.connect(src, op, dst)
        self.tb.run()

        # Check the data
        self.assertEqual(expected_results, dst.data())

        # Check the tags      
        self.assertEqual( expected_tag_offsets, [ x.offset for x in dst.tags()])
        
```

Which, of course, fails in the way this issue discusses:
```
43: AssertionError: Lists differ: [0, 1, 1, 2, 3] != [1, 1, 2, 3, 3]
43: 
43: First differing element 0:
43: 0
43: 1
43: 
43: - [0, 1, 1, 2, 3]
43: ?  ---
43: 
43: + [1, 1, 2, 3, 3]
43: ?           +++
43: 
43: 
```

(Yes this is a very old issue, but hey - still valid on master it seems?)

PR #4863 (2021-07-19): gr-blocks: (un)pack-k-bits tag propagation Bugfix #1111
fixes #1111
commit 6529fb50 (2021-07-19): digital: pack_k_bits propagate tags correctly

fixes #1111
closed 2021-07-19 by mormj