gnuradio/gnuradio #1122: grc: duplicate graph

osh, 2016-11-29:
There should be a button in GRC to duplicate the current open graph as a new one (unsaved/unnamed)
or also a "save copy as" in the file menu :+1:

jmcorgan, 2016-11-29:
'Save copy as' in menu would be preferred over a button that uses up screen real-estate.  Might consider automatically changing the option block id though to avoid duplicating/overwriting python script names.

osh, 2016-11-29:
a duplicate current graph option in menu would be super nice, with out without icon using valuable real-estate on low res screens

skoslowski, 2016-11-30:
Good idea(s) and should be fairly simple to implement. I agree with main menu only. And with that, I can't really see a strong not to add both, save copy as and duplicate. Should be fairly easy, too.

kartikp1995, 2017-01-20:
Hi,
I was looking into this issue. I believe there's already a `Save As` in file menu. Is there something I am missing?
Also, if we are planning to add a `duplicate` button, we will have to create a new `untitled.grc` with new id  `id_duplicate`. And open it in another tab in same `grc`. Is that correct?

I would like to look into this if someone else is not working on this.

Thanks!

skoslowski, 2017-01-20:
"Save copy as" usually means to save the current FG under a different filename, but keep the current filename -- sort of like a backup copy.
"Duplicate" just opens a new tab, then clones the current one into it, but does not setting a current filename.

I don't see a need for, nor would I want to have, extra/new metadata in either case. All the logic is there already. All in gui/Actions.py gui/Bars.py gui/ActionHandler.py

Didn't realize, that I have been assigned to this. So, yeah, go for it!

skoslowski, 2017-01-20:
Seems I can't set you as assignee, myself...

kartikp1995, 2017-01-24:
Thank you for your response.

But I am still confused between `Save As` and `Save Copy As`. How are those different?

skoslowski, 2017-01-24:
Jap, exactly.

kartikp1995, 2017-04-19:
Can be closed. #1188 is merged.

closed 2017-04-19 by jmcorgan