gnuradio/gnuradio #1131: FM Demod not working as expected

hedgesjack, 2016-12-06:
Version 3.7.10.1 - The 'FM Demod' block does not work as expected, unknown cause, I don't have the experience to look into source for a fix.

Verified that it is this block by using two development systems with older version 3.7.8 and 3.7.2, these both work as expected.

Signals appear to have phase error, easily seen with demodulating square wave data above 1kHz (previously modulated with 'Frequency Mod').

jmcorgan, 2016-12-06:
You'll have to be more descriptive in what you expected to happen that didn't, or what happened that you didn't expect.  Perhaps some working vs. not working plots as well (use an imgur type service).

hedgesjack, 2016-12-06:
Apologies I am fairly new to gnuradio and its workings...

I have a small GRC file that simply takes a square wave of 10kHz sampled at 384kHz, and modulates it with a sensitivity of 1 using the 'Frequency Mod' block. The data/signal is not altered in any way, simply 'looped back' directly to the 'FM Demod' block with the following settings;
Channel Rate: 384k
Audio Decimation: 1
Audio Pass: 180k
Audio Stop: 190k
Gain: 1
Tau: 75u

From the demodulated data, with our given data rates, you should see a clear square wave up to its 9th harmonic, and this is verified with the two working versions above. 
With the latest version, the output is a triangular waveform at a positive offset when a 10kHz or above square wave is used. With signals at low frequencies (below 1kHz) the demodulated data would look like a square wave with capacitive interference, as you increase the frequency, the amplitude reduces and an offset becomes apparent, clearly showing the demodulation process is somehow bugged.

At present I do not have an imgur account, however I shall try and get some visual representation of what I am experiencing from both the working version and 'bugged' version, that is if my explanation is not clear enough.

No external hardware is used, so it has been narrowed down to a gnuradio software bug.

jmcorgan, 2016-12-06:
You could attach the GRC file here. I'm not sure how soon we can look at it but that would let us replicate the problem.

hedgesjack, 2016-12-06:
It says the GRC file is unsupported, so here is a couple of screen captures, one of the block diagram (yes it went to something as simple as that to find the error!) and the other is the unexpected output. I presume you would know what a square wave with finite harmonics looks like.

I can use the older versions, however the development platform is incredibly slow that has it installed!
I shall try and get a screen capture of what is expected.

Thank you for your assistance.

[image]
[image]

hedgesjack, 2016-12-07:
Here is what should be recovered by the 'FM Demod' block, this is taken from version 3.7.8 and cannot be replicated in 3.7.10.1.

[image]

jmcorgan, 2016-12-07:
I'm pretty sure I know what is happening here, but was waiting until I got back to the lab to test it out.

In short, the FM Demod block has a built-in deemphasis filter that is low pass filtering your demodulated signal.  There was a bug in this block in older versions of GR that caused this filter to be disabled, so you didn't see this before, but the bug was fixed in 3.7.10, so now you see its effects.

Try demodulating with the Quad Demod block and see if this works for you.

hedgesjack, 2016-12-07:
That makes sense, thank you. Is there a bug in the filter still or is it simply not fit for what I am trying to use it for, just for reference?

Using the Quad Demod block works fine for us.

Once again thank you for your assistance.

jmcorgan, 2016-12-08:
It's not the right block in the first place, though it is poorly named and you can't be faulted for using it.

Pure "FM demodulation" wouldn't use a filter.  The deemphasis filter in the "FM Demod" block is really intended for use with NBFM analog modulation where there is a corresponding preemphasis filter on the transmitter. We should probably rename it to "NBFM Demod".

In any case, since the quadrature demod block is working for you, I'll close this.

closed 2016-12-08 by jmcorgan