gnuradio/gnuradio #1144: header_format_base::format comments appear to be misleading

jblandrum, 2016-12-13:
I'm working on an OOT module and I'm reading the header_format_default code and I think the comments in the base class for the function state that the output vector is a vector that includes the payload with the header prepended. When you look at the header_format_default as an example on how to inherit it it appears that rather this is a vector that only includes the header TO be prepended later on in the flowgraph. 
Comments imply output vector = header+payload
Reality seems to be output vector should = header

jmcorgan, 2016-12-13:
Can you put links to the specific lines of code you are referring to above in each?

jblandrum, 2016-12-14:
Yes I suppose that would have been helpful!

The following link makes the param seem to be something that has the entire payload prepended with the header.
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/include/gnuradio/digital/header_format_base.h#L141-L142

This implementation appears to be crafting only a header and putting that in the output and doesn't handle the input param at all.
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/header_format_default.cc#L103-L123

jmcorgan, 2017-01-26:
Agree with your assessment.  Can you submit a PR with the change to the documentation against the maint branch?

closed 2017-02-27 by jmcorgan