gnuradio/gnuradio #1153: Python 3?

fake-name, 2017-01-05:
Since python 2 is creeping closer and closer to EOL (just 3 years now!), does GNURadio support/have-plans-to-support Python3?

All my personal projects for the last 2+ years have been Py3K only, and it'd be nice to not have to install py2k just for gnuradio.

skoslowski, 2017-01-05:
Its in the works. GR 3.8 will support python3 (and keep legacy python support). Checkout https://www.youtube.com/watch?v=BM4L6wj60qM

jmcorgan, 2017-01-05:
Yeah, it's "in progress." We'll be doing an announcement in the next couple weeks on 3.8 status.

fake-name, 2017-01-05:
Awesomesauce!

The only comment I'd have then is it'd be nice to have that noted somewhere, unless I just missed the note?

jmcorgan, 2017-01-05:
We (GNU Radio team) is working a formal roadmap document for that announcement, so stay tuned.

closed 2017-01-05 by jmcorgan

Juanlu001, 2017-11-15:
I see #1467 and #1435. Can you share more information about this? These days [NumPy announced that they will drop Python 2 in 2019](https://github.com/numpy/numpy/blob/master/doc/neps/dropping-python2.7-proposal.rst).