gnuradio/gnuradio #1155: GNU Radio Companion assumes executing python files leads to them being run in the interpreter.

fake-name, 2017-01-05:
This is a minor issue, but right now, it appears that the GNU Radio Companion assumes that the "open-with" application for python files is set to `python.exe` (on windows).

For example, I have python files set to open with sublime text. As a result, some of the menu options in GRC don't work. Going to `Tools` -> `Filter Design Tool` just opens a script file.

I think this is probably something as simple as accidentally calling `subprocess.call(["gr_filter_design.py"])`, rather then the more specific `subprocess.call(["python", "gr_filter_design.py"])`, but it can be confusing if you don't know what's going on.

skoslowski, 2017-01-10:
You're right, it basically is (should be) a `subprocess.call()` and it is not explicitly using an interpreter in the call args. This way, there is no assumption on how `gr_filter_design` is implemented. 

I would prefer to have the (windows) installer generate a proper entry point for `gr_filter_design`, rather than adding logic on how to call that app on various platforms from GRC. That is how `setuptools` does it.

fake-name, 2017-01-10:
Yeah, I don't really know the best way to resolve this. Maybe use a `.bat`/`.sh` file shim to call the correct executable/script depending on platform?

jmcorgan, 2017-01-10:
If this is a Windows only problem, I'd defer to @gnieboer to solve for that installer, as Windows support for GNU Radio is maintained by him.

gnieboer, 2017-01-11:
I will take a look.  The run_GRC.bat file essentially does what you described, so it'd be a simple rename to drop another shim in and ensure it's running on the expected python environment.

gnieboer, 2017-01-11:
And BTW, this issue is limited to the filter design, all the other GRC calls to run python scripts use the specify to use the currently running executable.

fake-name, 2017-01-11:
FWIW, I probably wouldn't have even bothered reporting the issue if it weren't one of the first-level menu options in GRC. It's pretty minor even as it is.

fake-name, 2017-01-17:
This appears to be true for a lot of the underlying scripts too. `gr_modtool` is actually a python script, and I bet a bunch of the other management scripts are as well.

Personally, I think the interpreter should be explicit in nearly every case, but that's my preference.

fake-name, 2017-01-17:
Ok, this is really way more annoying, actually. All the command line python scripts are broken.

For example, `gr_modtool` correctly resolves out to `C:\Program Files\GNURadio-3.7\bin\gr_modtool.py`, but `python gr_modtool` fails unless the current working directory is `C:\Program Files\GNURadio-3.7\bin`.

closed 2024-03-11 by marcusmueller