gnuradio/gnuradio #1156: (Harmless) error outputs while loading FPGA firmware on to USRP.

fake-name, 2017-01-05:
Working in GRC, it appears that when one uses a UHD USRP interface, there is an attempt to catch the "loading %xx" output from the USRP  interface, and draw a pretty progress bar showing the progress of firmware load.

However, it seems like either the way the loading dialog is drawn has changed (swapping `\b\b\b\b\b\b` for `\r`?), or it does not work on windows.

As such, when starting my script on windows, you get:
```
IndexError: string index out of range
100Traceback (most recent call last):
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\core\Messages.py", line 116, in send_verbose_exec
    send(verbose)
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\core\Messages.py", line 51, in send
    messenger(_indent + message)
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\MainWindow.py", line 235, in add_console_line
    self.text_display.insert(line)
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\Dialogs.py", line 63, in insert
    line = self._consume_backspaces(line)
  File "C:\Program Files\GNURadio-3.7\lib\site-packages\gnuradio\grc\gui\Dialogs.py", line 75, in _consume_backspaces
    while line[back_count] == '\b':
IndexError: string index out of range
```

spammed into the terminal for the 30 or so seconds it takes the USRP to configure.

This is on windows, and is basically a cosmetic issue.

skoslowski, 2017-01-05:
Already fixed. See https://github.com/gnuradio/gnuradio/issues/1043

closed 2017-01-05 by skoslowski

fake-name, 2017-01-05:
Whoops, sorry. Derp on my part.