gnuradio/gnuradio #1162: header_format_default assumes 1 bit per symbol

v1bri, 2017-01-12:
I was playing around with the `packet_loopback_hier.grc` example and noticed that switching the header to a QPSK constellation had no effect on the `payload_symbols` tag attached in the `header_format_default::header_payload()` method. It looks like the class is hard coded to assume a 1 bit per symbol constellation. Here's the offending line.
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/header_format_default.cc#L217

The `header_format_counter` class does the correct thing.
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/header_format_counter.cc#L111

A possible solution would be to add a `bps` parameter to the constructor and `make()` method, similar to `header_format_counter`. This would change the interface though we could give it a default argument perhaps? I'm willing to submit a pull request and open to suggestions.

PS: I think this is also an issue with `header_format_crc`.
https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/header_format_crc.cc#L154

PPS: The examples have been very helpful in figuring everything out. Great documentation too.

jmcorgan, 2017-01-26:
I think your suggestion is useful, can you submit a PR for master branch with the change and default argument?

v1bri, 2017-01-31:
Ping?

commit aabe4f82 (2017-03-21): digital: Add Bits per Symbol parameter to header_format_default

Added payload bits per symbol parameter to the header_format_default
block. This allows it to be used with payload modulations besides, for
example, BPSK.

Fixes #1162
PR #1190 (2017-03-21): digital: Add Bits per Symbol parameter to header_format_default
Added payload bits per symbol parameter to the header_format_default
block. This allows it to be used with payload modulations besides, for
example, BPSK.

Fixes #1162
closed 2017-03-21 by jmcorgan