gnuradio/gnuradio #1164: Oh god indentation disaster in all the source!

fake-name, 2017-01-12:
It appears the entire source tree has mixed indentation. Additionally, it seems that the tabs in the source are from someone who has their computer set to a tab-width of 8 spaces, independent of  the *actual* indentation-per-level of the code.

This means that an editor that infers the tab-stop from the source mis-indents the C++ code when it's opened (since the C++ seems to mostly use a 2-space indentation).

Would there be any interest in a whitespace cleanup? I'd be happy to put together a whitespace-only pull request that at least gets rid of the tabs.

jmcorgan, 2017-01-12:
Yeah, there are (especially in older code that hasn't been touched for a while) some indentation issues as you describe. For new code we tend to use the Emacs Python and C++ mode defaults, and strip extraneous whitespace.

However, whitespace change commits tend to break or make difficult to merge outstanding branches that are in progress, and we have a few big ones for the python3 development effort.  In fact, a lot of the Python indentation issues are being fixed there along the way.

It will be a little time before we get to the point where we can do this properly, and given the divergence of maint/master/next it has to be done one branch at a time.

I'll keep this open until we're ready, and, can you please edit the issue title to be less dramatic?

fake-name, 2017-01-12:
> can you please edit the issue title to be less dramatic?

Awww, I thought it was funny.

fake-name, 2017-01-13:
Would there be an interest whitespace cleanup of the C++ files only?

mbr0wn, 2017-03-26:
We even have a tool for this (`dtools/bin/remove_whitespace`), and we used to run it occasionally in the past. It only removed trailing whitespace, though.

fake-name, 2017-03-26:
I'd suggest something like developing a  [`astyle`](http://astyle.sourceforge.net/) ruleset that can be run against the whole tree. That could be part of the repository.

jmcorgan, 2017-03-26:
@fake-name Let me know if you develop this.  Otherwise, it's going to be pretty low on this of things right now.

fake-name, 2017-03-26:
Well, I mean, putting together a astyle-ruleset is quite easy (I've done it to 80+KLOC codebases in the past without issue), but the problem as I understood it was that a bunch of whitespace changes would break the some merges with a development branch?

jmcorgan, 2017-03-26:
Not yet, but it's getting closer.

fake-name, 2017-03-26:
The big challenge after that would be deciding a coding style. astyle can put your code in whatever format you want.

 - Indentation: Spaces? Tabs? 2-space or 4 space (3 space if you hate everything)
 - Bracket style: Allman? K&R? Whitesmith? (and there's a bunch more)
 - Do you indent the interior of classes?
 - And switch statements?
 - And labels?
 - What about preprocessor statements?
 - Do you want to add whitespace padding to operators?

Basically, you have to design a coding style, and astyle then puts your code into that format. 

See the options [here](http://astyle.sourceforge.net/astyle.html).

------

Of course, the **correct** style is tab-indentation, Allman brackets, indented class bodies, and whitespace padding \</snark\>

jmcorgan, 2017-03-26:
Almost all the C++ code in GNU Radio was originally written with Emacs C++ mode; with nothing else to go on, mimicking the same choices made there would get you somewhere acceptable.

fake-name, 2017-03-26:
Well, that's a good place to start!

bastibl, 2017-03-26:
Is there a reason to use astyle and not clang-format? AFAIK, the latter has a more robust parser.

fake-name, 2017-03-26:
Nope, other then I've used astyle.

jmcorgan, 2017-04-04:
@fake-name Do you plan to continue on this?

fake-name, 2017-04-04:
Whoops, haven't had time, and the project at work that involved gnuradio has moved on.

jmcorgan, 2017-04-04:
Ok, closing.

mhostetter, 2017-04-04:
@jmcorgan this is something I'd be interested in picking up moving forward. I like @bastibl idea of using clang-format. Would you like me to look into it?

jmcorgan, 2017-04-04:
Go ahead.