gnuradio/gnuradio #1166: grc: Virtual sink ignores message connections with duplicate port name

mhostetter, 2017-01-13:
I discovered a bug with a virtual sink with multiple message input connections from blocks with the same output port name.  GRC will only connect one message port and ignore the rest.

Here's an example flowgraph:
[image]

The output is:
[image]

The .grc and .py files are attached in the zip.  You can see in the GRC-generated .py file, only the first message strobe is connected to the debug print.

[test.zip](https://github.com/gnuradio/gnuradio/files/704847/test.zip)

mhostetter, 2017-01-13:
Upon further examination, even ports with unique names are ignored.  It appears GRC only uses the first connection when generating the python.

Adding an extra message port, shown below, will also get ignored in the .py file.
[image]

The modified .grc and .py are attached.

[test.zip](https://github.com/gnuradio/gnuradio/files/704871/test.zip)

skoslowski, 2017-01-13:
Virtual connections were never meant to handle multiple upstream blocks. I guess messages can after.
That feature has been on my "to-be-rewritten" for 3.8 for a while. Might as well start now...

This is the first half of it. Your test-cases runs as expected:
https://github.com/skoslowski/gnuradio/tree/fix_virtual_connection_1166

I might do another iteration on it before putting it in a PR. Feedback welcome, though.

jmcorgan, 2017-01-26:
Fixed in #1168

PR #1168 (2017-01-26): Virtual connection with multiple upstream connections (messages) and Wildcard pad
This fixes #1166. I went ahead and did a rewrite of the type/vlen resolution code, that was planned for 3.8. Additional/new features like resolving through bypassed ports will be continued on next.

Along the way I found that the "Wildcard" setting for pad block didn't work and fixed that, too.
closed 2017-01-26 by jmcorgan