gnuradio/gnuradio #1171: Ports accept bytes, but the block converts to char?

fake-name, 2017-01-17:
This is more of an annoyance then a legitimate issue, but there seems to be a consistent inconsistency in the port data types versus the block names for byte-types.

Basically, the type converter blocks that take or produce values of type `Float` are called `xxx to Float` or `Float to xxx`. The type converter blocks that take or produce values of type `Complex` are called `xxx to Complex` or `Complex to xxx`. This rule holds true for every block, *except* the byte converter blocks.

For byte converter blocks, the block that takes an input data of type `Byte` is called `Char to xxx` or `xxx to Char` (or `UChar`, for extra confusion).

This makes the discoverability of these blocks minimal, without some supplemental googling. <kbd>ctrl</kbd> + <kbd>f</kbd> \<datatype\> finds the converter blocks for all *other* types without needing to leave GRC, so I actually initially didn't think there *were* any byte conversion blocks until I did more research.

It'd be nice to pick one word for the 8-bit type, and stick with it throughout the UI. Right now, it's confusing.

mbr0wn, 2017-01-20:
We prefer if you file minor issues (assuming they're legit issues and you provide enough background info, which you are).

roman-popenov, 2020-12-05:
I just want to confirm that `char` is indeed is `byte` in this case. Not that I am a purist or know much how it works here and there, but impression was that
```text
The main difference between a byte and char data type is that byte is used to store raw binary data while other is used to store characters or text data. […] In terms of range, a byte variable can hold any value from -128 to 127 but a char variable can hold any value between 0 and 255.
```
What the nuances between `UChar` vs. regular `Char`? By the sound of it, it seems like GNU `UChar` is regular char and GNU `Char` is regular byte.

fake-name, 2020-12-06:
`UChar` and `Char` are unsigned character and character storage types respectively.

`Char` and `UChar` are defined by the C spec as having a `minimum` size of 8 bits. However, there is **not** a maximum size, so it's conceivable they could be larger. In general, they're 8 bits exactly, but this is implementation defined.

Really, if you want a explicitly 8-bit datatype, you  want to use `Int8_t` and `UInt8_t` from `stdint.h`. 

So functionally, there's two issues here:

 - Use of `char` as a insufficently qualified stand in for a explicitly-sized variable
 - Mixing of `char` and `byte` for the same data. *On this platform* (Clang/MSVC/GCC, most modern platforms), they are indeed the same, but this isn't guaranteed by the C spec.

roman-popenov, 2020-12-18:
How can I easily differentiate between a stream of potential `Char`s or `Bytes` for `File Source`? As far as I am concerned, I could be streaming chars or bytes in this case from a file. Is it completely based on context in this case?

willcode, 2020-12-18:
@fake-name `uint8_t` and `int8_t` may be more correct, but `char` has been successfully used as an 8-bit type for a long time in a lot of projects. Fixing up/modernizing typing sounds like a good PR as long as it's done carefully.

@roman-popenov yes, it's based on context. File Meta Source/Sink and SIGMF could be used to provide additional metadata.

willcode, 2020-12-18:
@fake-name see http://www.cplusplus.com/doc/tutorial/variables/ where (u)char is defined as one byte. While it does have the amusing qualification `at least 8 bits`, lots of things will break if we put GR on a computer with, say, 12-bit bytes.

closed 2020-12-18 by willcode