gnuradio/gnuradio #1173: `gr.interp_block` seems to drop message tags

fake-name, 2017-01-18:
With the following simple flow-graph:

[image]

If I use a integrated block (such as the "Repeat" block) in the place of the `interp2x`, I get the expected message tags from both tag debug interfaces.

However, if I use a subclass of the python `gr.interp_block` in the place of the  `interp2x` block (including the one  in the tests at https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/python/blocks/qa_block_gateway.py#L86-L99), only the first message tag is passed through the block.

With "Repeat x=2" block:

```
C:\Code\grc>python tag_dropper.py
gr::pagesize: no info; setting pagesize = 4096
Press Enter to quit:
----------------------------------------------------------------------
Tag Debug: Bit-Sliced
Input Stream: 00
  Offset: 0  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 0  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 20  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Bit-Sliced
Input Stream: 00
  Offset: 40  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 40  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Bit-Sliced
Input Stream: 00
  Offset: 80  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 60  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------
```

With the following block:

```
class interp2x(gr.interp_block):
    def __init__(self):
        gr.interp_block.__init__(
            self,
            name = "interp2x",
            in_sig = [numpy.int8],
            out_sig = [numpy.int8],
            interp = 2
        )

    def work(self, input_items, output_items):
        output_items[0] = input_items[0]
        return len(output_items[0])

```

```

C:\Code\grc>python tag_dropper.py
gr::pagesize: no info; setting pagesize = 4096
Press Enter to quit:
----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 0  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Bit-Sliced
Input Stream: 00
  Offset: 0  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 20  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 40  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 60  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------

----------------------------------------------------------------------
Tag Debug: Raw
Input Stream: 00
  Offset: 80  Source: n/a     Key: packet_len   Value: 20
----------------------------------------------------------------------
```

As you can see, only the first message tag ever makes it through the graph. If I set `interp = 1` in the `gr.interp_block.__init__()` call, it does pass the message tags, but that's useless to me as I need to be able to do interpolation.

Complete test script:

```
#!/usr/bin/env python2

from gnuradio import blocks
from gnuradio import eng_notation
from gnuradio import gr
from gnuradio.eng_option import eng_option
from gnuradio.filter import firdes
from optparse import OptionParser
import pmt
import numpy

TEST_INTERP2X = True

class interp2x(gr.interp_block):
    def __init__(self):
        gr.interp_block.__init__(
            self,
            name = "interp2x",
            in_sig = [numpy.int8],
            out_sig = [numpy.int8],
            interp = 2
        )

    def work(self, input_items, output_items):
        output_items[0] = input_items[0]
        return len(output_items[0])

class top_block(gr.top_block):

    def __init__(self):
        gr.top_block.__init__(self, "Top Block")

        ##################################################
        # Variables
        ##################################################
        self.samp_rate = samp_rate = 32000

        ##################################################
        # Blocks
        ##################################################
        self.blocks_message_strobe       = blocks.message_strobe(pmt.intern("TEST"), 2000)
        self.blocks_random_pdu           = blocks.random_pdu(20, 20, chr(0xff), 1)
        self.blocks_pdu_to_tagged_stream = blocks.pdu_to_tagged_stream(blocks.byte_t, 'packet_len')

        self.blocks_tag_debug_raw        = blocks.tag_debug(gr.sizeof_char*1, 'Raw', "")
        self.blocks_tag_debug_raw.set_display(True)

        if TEST_INTERP2X:
            self.blocks_repeat               = interp2x()
        else:
            self.blocks_repeat               = blocks.repeat(gr.sizeof_char*1, 2)

        self.blocks_tag_debug_repeated   = blocks.tag_debug(gr.sizeof_char*1, 'Bit-Sliced', "")
        self.blocks_tag_debug_repeated.set_display(True)

        self.blocks_throttle             = blocks.throttle(gr.sizeof_char*1, samp_rate,True)
        self.blocks_null_sink_0          = blocks.null_sink(gr.sizeof_char*1)

        ##################################################
        # Connections
        ##################################################
        self.msg_connect((self.blocks_message_strobe, 'strobe'), (self.blocks_random_pdu, 'generate'))
        self.msg_connect((self.blocks_random_pdu, 'pdus'), (self.blocks_pdu_to_tagged_stream, 'pdus'))
        self.connect((self.blocks_pdu_to_tagged_stream, 0), (self.blocks_repeat, 0))
        self.connect((self.blocks_pdu_to_tagged_stream, 0), (self.blocks_tag_debug_raw, 0))
        self.connect((self.blocks_repeat, 0), (self.blocks_tag_debug_repeated, 0))
        self.connect((self.blocks_repeat, 0), (self.blocks_throttle, 0))
        self.connect((self.blocks_throttle, 0), (self.blocks_null_sink_0, 0))

    def get_samp_rate(self):
        return self.samp_rate

    def set_samp_rate(self, samp_rate):
        self.samp_rate = samp_rate
        self.blocks_throttle.set_sample_rate(self.samp_rate)

def main(top_block_cls=top_block, options=None):

    tb = top_block_cls()
    tb.start()
    try:
        raw_input('Press Enter to quit: ')
    except EOFError:
        pass
    tb.stop()
    tb.wait()

if __name__ == '__main__':
    main()
```

fake-name, 2017-01-18:
If I manually copy the tags across to the output stream, I get the expected result:

```

        nread = self.nitems_read(0)
        startoffset = self.nitems_written(0)
        tags = self.get_tags_in_range(0, nread, nread+num_input_items)

        for tag in tags:

            tag_offset = max(((tag.offset - nread) * self.interp_factor) - 1, 1)
            output_offset = self.nitems_written(0) + tag_offset

            self.add_item_tag(0, output_offset, tag.key, tag.value)
```

I'd guess that the tag propigation isn't automatic on `gr.interp_block` instances, but it somehow is on the C++ interp blocks, and as I understand it, the python parent class is just a minimal wrapper over the C++ class.

jmcorgan, 2017-01-26:
Note that when writing Python work functions, the passed references are to an array of arrays, that is, input_items[0] is a reference to an array for input port 0, input_items[1] for port 1, etc., and the same for ouput_items.  To write output, you assign to a slice of the output array. 

See section 3.2.3 in:

http://gnuradio.org/redmine/projects/gnuradio/wiki/Guided_Tutorial_GNU_Radio_in_Python

It's quite likely the erroneous assignment to the port array as done the test function above is causing the issue you describe.

fake-name, 2017-01-27:
Oh, I see.

To be clean, let me describe what I think you're saying is happening, so you can confirm:

 - Base block receives new data
 - Base block creates new `out` array, 
 - Base block applies input tags to the appropriate places in output array
 - Base block calls my code
    - My block clobbers output array with internally generated array
 - Base block no longer has the tags?

Does this mean the tags are attached to the numpy array somehow?

marcusmueller, 2023-10-24:
We reworked the tag handling on rate-changing blocks since 2017, needs to be checked if still true