gnuradio/gnuradio #1181: AttributeError: 'module' object has no attribute

ZibiDz, 2017-01-22:
Good Morning!
 
I've created my C++ decimator block, everything works well so far with 
gr_modtool and "make && sudo make install && sudo ldconfig" till I add a fir filter object, like this:

`d_fir = new gr::filter::kernel::fir_filter_ccc(1, std::vector<gr_complex>(trainingssize, gr_complex(0,0)));`  .

GRC stops the flowgraph with:

Traceback (most recent call last):
  File "/home/zibi/Desktop/adaptiveFilter/GRC-Beispiele/mpsk.py", line 378, in <module>
    main()
  File "/home/zibi/Desktop/adaptiveFilter/GRC-Beispiele/mpsk.py", line 366, in main
    tb = top_block_cls()
  File "/home/zibi/Desktop/adaptiveFilter/GRC-Beispiele/mpsk.py", line 222, in __init__
    self.adaptive_filters_my_true_lms_dec_0 = adaptive_filters.my_true_lms_dec(0.2, (modulated_sync_word))
AttributeError: 'module' object has no attribute 'my_true_lms_dec'

>>> Done

I've already tried things I understood from similar threats as:
-checking install directory 
-export PYTHONPATH=/usr/lib/python2.7/dist-packages before gr_modtool
-remove and rebuild everything
-replacing FindBoost.cmake as Marcus advises caused an error

Ahh and "make" gives a "Warning 362: operator= ignored" in both cases, with and without the fir filters object 
Im on xubuntu 16.04 and gnuradio version 3.7.10.1 via installationsscript

Kind regards and thank you for your time!
Zibi

marcusmueller, 2017-01-23:
Hi Zibi, 

> as Marcus advises

 sounds like you're referring to [this](https://lists.gnu.org/archive/html/discuss-gnuradio/2017-01/msg00189.html) mailing list thread, is that correct?

I thought the system I was referring to there was an Arch Linux system, so I don't know whether this applies here at all:

>  Im on xubuntu 16.04 and gnuradio version 3.7.10.1 via installationsscript

What's that `installationsscript`?

You'd probably want to explain *where* you've added that FIR line, it's not completely clear.

Generally, it's a good idea to have a minimal example of how to reproduce your problem locally; since you're already on github, why not make a git repo out of your OOT Module and push it to github and refer to it here?

mbr0wn, 2017-01-23:
@ZibiDz this doesn't look like a bug in GNU Radio, so I'm closing this for now. Please take the discussion to the list, and if it does turn out to be a GNU Radio bug, we'll reopen this.

closed 2017-01-23 by mbr0wn

ZibiDz, 2017-01-23:
Hello Marcus,

I tried to replace the FindBoost.cmake as you adviced to Jacqueline.. but
this causes some errors:

-- Build type not specified: defaulting to release.
CMake Error: Error in cmake code at
/usr/share/cmake-3.5/Modules/FindBoost.cmake:5:
Parse error.  Expected a command name, got unquoted argument with text
"<!DOCTYPE".
CMake Error at CMakeLists.txt:89 (find_package):
  find_package Error reading CMake code from
  "/usr/share/cmake-3.5/Modules/FindBoost.cmake".

As Im using xubuntu and not Arch my FindBoost.cmake should work. (Again..
it compiles my code, the problem appears when I run the flowgraph)

I mean the
"$ wget http://www.sbrac.org/files/build-gnuradio && chmod a+x
build-gnuradio && ./build-gnuradio"how I installed gnuradio.

Im an electronic engineering student and not very familiar with programming
objective on computers, so I suppose it's not a big thing I did wrong.

in *impl.h:
     private:
      // Nothing to declare in this block.
     kernel::fir_filter_ccc *d_fir;

in*impl.cc in
    /*
     * The private constructor
     */
...
      d_fir = new kernel::fir_filter_ccc(1,
std::vector<gr_complex>(trainingssize, gr_complex(0,0)));

it's taken from hilbert_fc_impl.cc and .h .

My *impl.h and *impl.cc files in the appendix.

marcusmueller, 2017-01-23:
As I said, the advice given to Jacqueline doesn't apply to you, and as Martin said, this is most probably not a bug in GNU Radio but in your OOT module, so please take this to the mailing list.

n-west, 2017-01-24:
This is a bug in your OOT because you aren't properly linking to libgnuradio-fft. Add FFT to the list of GR_REQUIRED_COMPONENTS in your module's top level cmakelists

KyleHerndon, 2018-06-04:
I am having the same problem as Zibi. I have found numerous instances of this error documented around the internet, but no solution. Martin claims its not an issue with GNURadio, but I am under the impression that most of the people who have this issue are like me, meaning we followed the tutorial on how to make a new mod using gr_modtool and encountered this error. My intuition is that any linking error should have been solved when I used cmake. My module also doesn't require any components from other libgnuradio files, and thus definitely shouldn't require editing the top level cmakelists.

You can find my (very simple) attempt at a OOT module at: https://github.com/KyleHerndon/SDR-gnuradio/tree/master/gr-pipes

I'm on Ubuntu 16.04 and GNU Radio Companion 3.7.11.1 installed by
`sudo apt install gnuradio`

Any direction would be greatly appreciated.