gnuradio/gnuradio #1182: map frequencies to numbers

St0ner1995, 2017-01-23:
Hello, is it possible to get a module that allows me to map frequencies (or any number) to a number (like a channel id)

my goal is to create a software CB Radio and have all 80 channels (I live in Australia, don't know if this is the same everywhere) mapped to their frequencies

if you look [here](http://uhfcb.com.au/80-Channel-UHF-Information.php) you will see the channel frequencies and spacing (not bandwidth) that is used over here. it is not as simple as doing some simple maths as the channels are set out in this order, 1,41,2,42,3,43,4,44 etc...

marcusmueller, 2017-01-23:
Hi @St0ner1995!

You can certainly do that, but it's *very* application-specific, and also not really something that you'd need a GNU Radio module for (especially not inside the main source tree) – both Python and C++ offer mapping types, `dict` in Python or `std::map<keytype,valuetype>` in C++.

mbr0wn, 2017-01-23:
@St0ner1995 what @marcusmueller said, and also, there's some applications that already do this (like with ARFCNs and GSM). Since this is not an issue with core GNU Radio, I'm closing this. I would recommend taking this to discuss-gnuradio if you have more questions.

closed 2017-01-23 by mbr0wn