gnuradio/gnuradio #1184: GRC right click "Create Hier" fails

giladbeeri, 2017-01-24:
I have a regular flowgraph in GRC 3.7.10.1 (macOS 10.12.1). I wanted to follow the instructions in https://lists.gnu.org/archive/html/discuss-gnuradio/2014-02/msg00024.html.

1. I selected few blocks, right clicked, More -> Create Hier.
2. GUI effect: nothing happens, no indication of success or failure
3. Console: shows the exception which is probably the reason of failure

```
Generating: '/Users/XXX/.grc_gnuradio/flowgraph.py'
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/grc/gui/ActionHandler.py", line 238, in _handle_action
    for connection in block.connections:
AttributeError: 'Block' object has no attribute 'connections'
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/grc/gui/ActionHandler.py", line 238, in _handle_action
    for connection in block.connections:
AttributeError: 'Block' object has no attribute 'connections'
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/grc/gui/ActionHandler.py", line 238, in _handle_action
    for connection in block.connections:
AttributeError: 'Block' object has no attribute 'connections'
Traceback (most recent call last):
  File "/opt/local/Library/Frameworks/Python.framework/Versions/2.7/lib/python2.7/site-packages/gnuradio/grc/gui/ActionHandler.py", line 238, in _handle_action
    for connection in block.connections:
AttributeError: 'Block' object has no attribute 'connections'
```

skoslowski, 2017-01-24:
Looks like auto-refactoring gone wrong.... 11 Months ago =)

@sdh11, I recall you wanting to this for 3.8. Would it make sense to backport your version?

garengllc, 2017-10-31:
Seems to be a problem in 3.7.12git-197-gbd611224 still.

nickoe, 2019-02-23:
This seems to work on current master, probably fixed by #1752. Maybe this needs to be closed, or maybe the 3.8 milestone needs to be removed for this if a fix is pending for the maint branch.

closed 2019-02-25 by noc0lour