gnuradio/gnuradio #1185: Qt time sink complains about template syntax

primercuervo, 2017-01-25:
I just pulled the latest master (6225e5d4d1edd67c896c6c43bdc584a078480a8d) and I found an unusual behaviour with the qt time sink. I'm working on an Ubuntu 14.04 machine but also got to reproduce it in a Ubuntu16.04 one.

To reproduce it, I just generate a flowgraph with signal source -> throttle -> qt time sink. I don't modify any of their parameters. After clicking run, I get:
```

Executing: /usr/bin/python2 -u /home/user/temp/top_block.py

  File "/home/user/temp/top_block.py", line 67
    self.qtgui_time_sink_x_0 = Template error: #set $win = 'self._%s_win'%$id
                                            ^
SyntaxError: invalid syntax

>>> Done (return code 1)
```

primercuervo, 2017-01-25:
For now I reverted to cb68a76c4ae7c97efcc9e6c48843c2f5f0cf5d84 where it is working fine.

kartikp1995, 2017-01-25:
Hi,

I will look into it. 

Thanks for the notification.

jmcorgan, 2017-01-25:
We are right on the cusp of a release.  If I don't get a tested fix in the next day or so I will have to revert #1150 and catch the fix on the next release.

jmcorgan, 2017-01-26:
I'll get #1187 into the maint and master releases, thanks.

primercuervo, 2017-01-26:
Great! just tested the time sink and it seems happy again. Thank you for your time on this!

jmcorgan, 2017-01-26:
Fixed in #1187

closed 2017-01-26 by jmcorgan