gnuradio/gnuradio #1186: gqrx works with HackRF board but not gnuradio-companion: FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB

SonnyRajagopalan, 2017-01-25:
I raised this issue with hackrf github, but perhaps it is more appropriate here. I have a hackrf board and am able to use it with `gqrx`. However, when I try to execute a flow graph I created with gnuradio-companion, it fails for the following reason:

    Generating: '/mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py'

    Executing: /usr/bin/python2 -u /mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py

    linux; GNU C++ version 4.8.4; Boost_105400; UHD_3.11.0.git-59-g52ca2e0b

    gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
    built-in source types: file osmosdr fcd rtl rtl_tcp uhd miri hackrf bladerf rfspace airspy soapy redpitaya
    Number of USB devices: 13
    USB device 1d50:0089: 0000000000000000909864c82c3a2fcf match
    Using HackRF One with firmware 2015.07.2
    gr-osmosdr v0.1.x-xxx-xunknown (0.1.5git) gnuradio 3.7.10
    built-in sink types: uhd hackrf bladerf soapy redpitaya file
    Number of USB devices: 13
    USB device 1d50:0089: 0000000000000000909864c82c3a2fcf match

    FATAL: Failed to open HackRF device (-1000) HACKRF_ERROR_LIBUSB

    Trying to fill up 1 missing channel(s) with null sink(s).
    This is being done to prevent the application from crashing
    due to gnuradio bug #528.

    Traceback (most recent call last):
      File "/mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py", line 128, in <module>
        main()
      File "/mymachine/sdr/gnuradioProjects/flowGraphsFromGRC/top_block.py", line 117, in main
        tb.start()
      File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/top_block.py", line 109, in start
        top_block_start_unlocked(self._impl, max_noutput_items)
      File "/usr/local/lib/python2.7/dist-packages/gnuradio/gr/runtime_swig.py", line 4879, in     top_block_start_unlocked
    return _runtime_swig.top_block_start_unlocked(*args, **kwargs)
    RuntimeError: list contains invalid format!

Any insight is appreciated. Thanks!

jmcorgan, 2017-01-25:
The issue with the GNU Radio bug has been corrected in more recent GNU Radio versions.  Please update to the latest maintenance release, 3.7.10.1.

However, it appears you have permissions issues with your HackRF device resulting in it's failed initialization.

As this is a forum for reporting bugs in GNU Radio, I suggest you take this to the GNU Radio mailing list instead; you'll generally find help fairly quickly.

closed 2017-01-25 by jmcorgan

SonnyRajagopalan, 2017-01-25:
I don't think the latest version will fix this, I am running v3.7.10.1-182-g59136936. I do not have a permissions issue with LIBUSB as gqrx works just fine with the same user and hackrf board.