gnuradio/gnuradio #1199: gr-uhd: out of order message based timed commands

mrjacobagilbert, 2017-02-02:
I am have run into a niche issue, and wanted to discuss whether or not this is even desired to be addressed in-tree.

Issue: when using timed message-based commands to tune USRP devices, it is possible that a timed command intended for the TX side can inadvertently end up queued before a later RX timed tune command (and vice versa). Because messages are asynchronous, it's not guaranteed that a block emitting such tuning commands will be able to use emission order to correctly get these commands to the USRP. This can cause an earlier tune to be held up until a late tune on the other side is handled.

Currently any tune command is applied to the side of the USRP corresponding to the type of block that receives the message (sync / source) and I do not see a way around this. I have a couple methods of dealing with this, but all require nontrivial changes to gr-uhd. Is addressing this something that would be desired? Is anyone aware of a way to around this with the current codebase?

mrjacobagilbert, 2017-02-04:
If there aren't any strong feelings about this from the Ettus side, I will submit a PR that does the following:

1) When a message is received into either the USRP Source / USRP Sink it will check to see if the command has a 'direction' key. If it does not, it will default to whatever type the block is (source -> rx, sink -> tx)

2) Messages that can apply to TX or RX will attempt to be applied accordingly. If a TX command is received without a sink block present it will do nothing, likewise for RX.

This should remain API compatible, add minimal overhead, and allow for deterministic control of tuning for transceiver applications.

mbr0wn, 2017-02-09:
@mrjacobagilbert yep PRs are welcome. What you suggests sounds sensible.

mrjacobagilbert, 2017-08-13:
I apologize for letting this sit for so long. I have a fix here but need to test w/ hardware early next week as it is slightly different (and less confusing/complicated) than what I have been using successfully.

https://github.com/mrjacobagilbert/gnuradio/tree/uhd_msg_tune_determinism_fix

devnulling, 2020-01-21:
Closing as this was fixed with https://github.com/gnuradio/gnuradio/pull/1422

closed 2020-01-21 by devnulling