gnuradio/gnuradio #1203: http://pybombs.info should simply directly redirect to Github pybombs project

marcusmueller, 2017-02-04:
also, for some reason, when I click on that same link, I get a blank page instead of the linked github repo. I think this happens because github doesn't like to live in a HTML `<frame>` (which is what `pybombs.info` is, the redmine wiki in a frame)

mbr0wn, 2017-02-09:
lol I had no clue we have pybombs.info.

duggabe, 2020-07-28:
I just checked the http://pybombs.info URL and got a 404. shouldn't this issue be closed?

777arc, 2020-07-29:
It's still an active domain, it's just not configured properly.

osh, 2020-07-29:
this has now been updated to point to github, best of luck

dkozel, 2020-07-31:
@osh Thanks for making the update. Firefox blocks the iframe embed. Can you switch it to use an HTML redirect?

```
<meta` http-equiv="refresh" content="0; URL=https://github.com/gnuradio/pybombs'" />
```

marcusmueller, 2020-10-30:
Pinging the timmiest of Tims on a side-channel. Bringing in @argilo

mormj, 2021-10-04:
Closing - resubmitted on pybombs issue tracker

closed 2021-10-04 by mormj