gnuradio/gnuradio #1204: variable maybe not initialized

maddin200, 2017-02-06:
gr-audio\lib\osx\osx_impl.cc line 46
char format_id[sizeof(asbd.mFormatID)+1];
better:
char format_id[sizeof(asbd.mFormatID)+1] = {0};

jmcorgan, 2017-04-04:
@michaelld Let me know if this is an issue or if it should be closed.

bastibl, 2017-04-04:
I think at least the `memcpy` arguments should be swapped. If I'm not mistaken it copies the uninitialized array into the variable.
https://github.com/gnuradio/gnuradio/blob/master/gr-audio/lib/osx/osx_impl.cc#L46

michaelld, 2017-04-04:
I think @bastibl is correct, that we need to swap the first 2 arguments. Glad it's just for debug purposes & not critical runtime! I'll verify & do a PR.

mbr0wn, 2017-07-14:
Looks like @michaelld fixed, and we can close?

michaelld, 2017-07-14:
Yes! Someone please do close this issue. I'm not given this option.

closed 2017-07-14 by jmcorgan