gnuradio/gnuradio #1207: Correlation Estimator regression

Tim---, 2017-02-08:
After trying to use the Correlation Estimator without success, I looked at the examples provided to understand how to make it work.

Unfortunately, the [example flowgraph](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/examples/demod/test_corr_est.grc) seems to be out of sync with the last changes made to the [Correlation Estimator block](https://github.com/gnuradio/gnuradio/blob/master/gr-digital/lib/corr_est_cc_impl.cc), and is completely unusable because almost every point of the output gets tagged. 

Increasing the threshold makes it better, but there is still some correlations where the |corr|^2 curve clearly shows zero, which looks like a bug.

I attached the output of |corr|^2 with a threshold of 0.9999999 :

[image]

jmcorgan, 2017-02-08:
Unfortunately, I have seen this same regression, but haven't had time to track it down.  It's likely related to some major changes made with the corr_est block a few months ago.

jmcorgan, 2017-02-08:
@awalls-cx18

nowls, 2017-02-14:
It has to do with a change in how the detection threshold is calculated. It used to be that the user would specify a fractional threshold value, `threshold`. When the correlation output magnitude is greater than `threshold` * sum(abs(correlation_sequence)^2), a detection is declared and tags are added. The sum is the l2 ("el"-2) norm squared, i.e., the power of the sequence. Hence thresholding is done w.r.t. an absolute level.

At some point the meaning of `threshold` changed. The current implementation uses a constant false alarm rate (CFAR) detection method. In this case, `threshold` is converted into a probability of false alarm:

`d_pfa = -logf(1.0f-threshold)`

Now, a detection is declared when the correlation output magnitude is greater than 4*(d_pfa * sum(correlation_mag of surrounding samples)). Basically, the threshold is dynamically chosen based on the noise (and/or interference) in the correlation magnitude of multiple samples surrounding the peak. This dynamic thresholding is useful, but the implementation has a couple of problems:

* The "surrounding samples" in this case are the correlation magnitude values computed over `noutput_items` of the current call to `work`. I believe that the constructor sets up a fixed number of expected items, so theoretically the number of surrounding samples will always be the same. I haven't verified this behavior.
* The factor of 4 seems arbitrarily chosen.
* There is no documentation about how this all works.

This modification ended up changing the expected behavior of the block and probably should not have been introduced as a bugfix, since it may have broken people's flowgraphs.

I implemented a CFAR corr_est block based on this implementation and guidance from [here](https://www.mathworks.com/help/phased/examples/constant-false-alarm-rate-cfar-detection.html). It allows the user to explicitly set a PFA, number of guard samples, and number of samples to average over. I will see if I can contribute it in the next week or two.

awalls-cx18, 2017-02-14:
Yeah, before trondeau submitted the change he asked me to review.  I didn't have time to give him much feedback, and I didn't have time to test against preambles which didn't have good correlation properties (the kind I often use: 1,-1,1,-1,1,-1...).  Although describing the log() of the threshold input as a Pfa and noting the algorithm is an outlier detection algorithm were part of my feedback.

I've known corr_est's threshold input had changed for quite a while, and have just used an OOT version of corr_est, instead of dealing with a 5 9's threshold.

I'll ask the question "what is corr_est's functional correlation requirement?"  Pattern matching against a preamble sequence that has been properly AGC'ed/scaled on the input, or searching for correlations in low SNR/wideband signals?

jmcorgan, 2017-04-04:
@nowls @awalls-cx18 Can you two summarize what you think is the best path to addressing this?

awalls-cx18, 2017-04-05:
@jmcorgan @nowls : Make the correlation peak detector algorithm selectable: static relative linear threshold, dynamic relative logarithmic threshold.
The first option above is the old behavior of the block. The second option above is the new, current behavior of the block.
The first option works well enough for high SNR situations with preambles with poor correlation properties. (VHF Maritime datalinks such as DSC and AIS leap to mind).
The second option works well for low SNR situations with sequences with nice correlation properties. (barker and gold code sequences?)

Because, do you really want a separate block for the 2 variants of a correlation detector?

FWIW, GNURadio tends to have a separate block for each permutation of something.  There are 3 analog PLL blocks, that all simply differ in their output function.  There are 2 clock sync blocks, M&M and PFB clock sync, that both perform the same overall function when they really just have different interpolators and timing error detectors.

The constellation modulator and demodulator blocks are the better pattern in my option: selectable constellations to support the overall function of the block.
In that vein, here is my symbol sync block that performs the high level function of symbol synchronization, but lets the user select the TED algorithm to use, the slicer constellation to use (if needed), and the interpolator to use.  https://github.com/awalls-cx18/gr-nwr/blob/master/grc/nwr_symbol_sync_xx.xml

My $0.02

-Andy

nowls, 2017-04-11:
@jmcorgan @awalls-cx18 : I agree with Andy. The user should be able to choose between the two methods, i.e., absolute or dynamic thresholding. At this point I'm not sure which would be better to make the default, since users of this block may have already dealt with the pain of changing their applications to reflect the change. At minimum it should be documented. Additionally, I think it would be useful to choose how many samples over which to compute the average noise level at the output of the correlator.   I believe this is currently fixed to a factor of the size of the correlation sequence.

rubund, 2017-05-20:
> At this point I'm not sure which would be better to make the default, since users of this block may have already dealt with the pain of changing their applications to reflect the change.

I think users have mostly dealt with the pain by changing to using an OOT version of corr_est_cc which has the old behaviour. This is what I did some time ago, since the corr_est_cc felt broken - somewhat of a nightmare :). I have been using corr_est_cc from gr-ais since.

I therefore think it should be unproblematic to roll back these few lines so that old flow graphs will start working again..

jmcorgan, 2017-05-22:
Code up a test branch and submit a PR for review and testing, please.

btashton, 2018-04-06:
@jmcorgan @awalls-cx18 I am putting together a patch tonight to fix this, but one thing I am not clear on is what the method `threshold()` should be returning in the dynamic peak detector is being used.  Right now it returns `d_thresh` which is clearly wrong.  Perhaps it should return `d_pfa`?

btashton, 2018-04-06:
@jmcorgan this is my first contribution to this project, so feedback is appreciated.

bhilburn, 2018-04-12:
@marcusmueller - Ping on this one.

btashton, 2018-04-12:
I ran some tests last night, and it seems to be working, but I do not have the enum parameter correct in the XML, it  shows up as a drop down, but the rendered code does not include the parameter.   I suspect it has to do with where the enum is namespaced.

marcusmueller, 2018-04-28:
I think you're on the right track with that; it's not the first enum we wrap into Python, but I constantly keep forgetting how to do that "correctly". I'll research that!

PR #1725 (2018-06-25): Restore ability to use absolute thresholding with corr_est
This extends the corr_est block to take a threshold method as an argument.

* THRESHOLD_ABSOLUTE restores the previous function of this block.
* THRESHOLD_DYNAMIC provides the outlier detection algorithm that was later added.

This should resolve #1207
closed 2018-06-25 by marcusmueller