gnuradio/gnuradio #1208: gnuradio-companion has bitrot on OS X?

buzmeg, 2017-02-09:
I've been trying to build gnuradio-companion on OS X.  I'm having a devil of a time handling all the dependencies.

Homebrew only seems to support QT5 while gnuradio wants QT4.  Also, homebrew isn't supporting the older pygtk/gtk 2.X combo that gnuradio wants.

Is there a different/maintained tool or build that I am missing?

Thanks.

jmcorgan, 2017-02-09:
You can build our next branch with QT5, though this branch is subject to rapid changes and is not considered stable.  We'll be supporting gtk3 on the next branch in the near future.

buzmeg, 2017-02-11:
How do I enable building with QT5?  I'm sort of in a circular dependency issue.  I grabbed next with:

git clone --branch next --recursive https://github.com/gnuradio/gnuradio.git

and then tried to run cmake with:

cmake -DENABLE_DOXYGEN=OFF -DCMAKE_PREFIX_PATH=$(brew --prefix qt5) -DCMAKE_INSTALL_PREFIX=/Users/andrewl/gnuradio ../

qtgui wants PyQT5 which is Python3 only but VOLK wants Cheetah templates which are Python2 only.

How do I break the chain?

Thanks.

bastibl, 2017-02-11:
Homebrew is pretty broken at the moment: https://github.com/gnuradio/gnuradio/issues/1090#issuecomment-278580591

You first have to install pyqt5 with python2 bindings
```
brew install --with-python pyqt5
```
This installs pyqt5 linked against qt@5.7 (brew info pyqt5), because the python bindings for qt5.8 are not ready yet. Make sense, but you also need qwt to build GNU Radio. This is, however, already linked against qt5.8 (brew info qwt). https://github.com/Homebrew/homebrew-core/commit/d2d15e863109fd2d0fdc24bb754166ba8f4e16e9

I manually installed the older version of qwt, which also links against qt@5.7
```
brew install https://raw.githubusercontent.com/Homebrew/homebrew-core/759d6680f83d223c6b086806862b22e1079af90a/Formula/qwt.rb
```

Make sure that you only have qt@5.7 installed and not qt (which is 5.8)

d13-kevinf, 2017-02-13:
Try this out, let me know if it helps you out. 
https://github.com/d13-kevinf/homebrew-core/tree/SierraGnuradioQT4Fosphor

https://github.com/d13-kevinf/homebrew-core/commit/edde746b7fac48fa1025239f9ccf933ef584d920

bastibl, 2017-03-26:
Homebrew updated pyqt5 to use the default qt5 libraries (as opposed to pulling in a second outdated version, which made compiling GNU Radio impossible). The `next` branch of GNU Radio is now straightforward to compile. You only have to enable the Python2 bindings for pyqt5 manually:

```
brew install --with-python pyqt5
```

jmcorgan, 2017-04-04:
Closing until OP reports back.

closed 2017-04-04 by jmcorgan