gnuradio/gnuradio #1209: Linear scaling in ESNO arg to gen_met, should be dB

jaredd, 2017-02-11:
In gr-fec, viterbi/metrics.c the parameter comments and subsequent code both express the esno scaling:
`double esn0,		/* Es/N0 ratio in dB */`[(link)](https://github.com/gnuradio/gnuradio/blob/master/gr-fec/lib/viterbi/metrics.c#L61)

but anywhere gen_met is called the esno is put into linear scale prior to the call:
[decode.cc:51](https://github.com/gnuradio/gnuradio/blob/da0a102e9b076db83fe5a63b3730ac39b588eb92/gr-fec/lib/viterbi/decode.cc#L51) and [decode_ccsds_27_fb_impl.cc:46](https://github.com/gnuradio/gnuradio/blob/da0a102e9b076db83fe5a63b3730ac39b588eb92/gr-fec/lib/decode_ccsds_27_fb_impl.cc#L46).

This likely leads to performance degradation v theory.

noc0lour, 2018-02-05:
Actually the code in [decode_ccsds_27_fb_impl.cc:46](https://github.com/gnuradio/gnuradio/blob/da0a102e9b076db83fe5a63b3730ac39b588eb92/gr-fec/lib/decode_ccsds_27_fb_impl.cc#L46) looks like it converts the value to linear scale before passing it to `gen_met` and then `gen_met` again converts the linear value to linear scale, which doesn't sound right, so I'd consider this a bug in the ccsds code. Nevertheless there seems no reason (other than practicality) to take in a `E_s/N_0` in dB scale.

marcusmueller, 2023-12-20:
looks like a persisting bug

marcusmueller, 2024-03-11:
very many things strange to broken about gr-fec's viterbi