gnuradio/gnuradio #1210: Possible resource leak: socket not closed in tuntap_pdu_impl.cc, set_mtu

jaredd, 2017-02-11:
In gr-blocks at [tuntap_pdu_impl.cc](https://github.com/gnuradio/gnuradio/blob/master/gr-blocks/lib/tuntap_pdu_impl.cc#L161) a socket is created but not closed before the function exits.  Recommend add close(sfd) before every return call.

k6nick, 2017-10-18:
I believe the socket should be closed only at the error code, not when the socket is returned without an error. Basically this:

      /* try to set MTU */
      if ((err = ioctl(sfd, SIOCSIFMTU, (void *) &ifr)) < 0)
        /* add this
           close(fd);
        */
        return err;

Simple enough fix, but I am not familiar at all with what's required to make this change. First time here, first time looking into the GNU radio code...

k6nick, 2017-10-18:
I made the fix and this is the pull request: https://github.com/k6nick/gnuradio/pull/1

I hope I didn't mess up anything, I'm just getting started with github.

marcusmueller, 2020-10-30:
> Recommend add `close(sfd)` before every return call.

That's been implemented, far as I can tell!

closed 2020-10-30 by marcusmueller