gnuradio/gnuradio #1212: Cross compiling fails

BradleyConn, 2017-02-15:
When trying to cross compile using the cmake file oe-sdk_cross.cmake, lines 4 and 5 fail because not enough arguments are given.

The lines are

`
string(REGEX MATCH "sysroots/([a-zA-Z0-9]+)" CMAKE_SYSTEM_PROCESSOR $ENV{SDKTARGETSYSROOT})
string(REGEX REPLACE "sysroots/" "" CMAKE_SYSTEM_PROCESSOR ${CMAKE_SYSTEM_PROCESSOR})`

Below is a partial excerpt of the cmake output

-- The CXX compiler identification is GNU 4.8.4
-- The C compiler identification is GNU 4.8.4
-- Check for working CXX compiler: /usr/bin/c++
CMake Error at /home/b/cross-compile/gnuradio/cmake/Toolchains/oe-sdk_cross.cmake:4 (string):
  string sub-command REGEX, mode MATCH needs at least 5 arguments total to
  command.
Call Stack (most recent call first):
  /home/b/cross-compile/gnuradio/build/CMakeFiles/3.2.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:2 (project)

CMake Error at /home/b/cross-compile/gnuradio/cmake/Toolchains/oe-sdk_cross.cmake:5 (string):
  string sub-command REGEX, mode REPLACE needs at least 6 arguments total to
  command.
Call Stack (most recent call first):
  /home/b/cross-compile/gnuradio/build/CMakeFiles/3.2.2/CMakeSystem.cmake:6 (include)
  CMakeLists.txt:2 (project)

CMake Error: Internal CMake error, TryCompile configure of cmake failed
-- Check for working CXX compiler: /usr/bin/c++ -- broken
CMake Error at /usr/share/cmake-3.2/Modules/CMakeTestCXXCompiler.cmake:54 (message):
  The C++ compiler "/usr/bin/c++" is not able to compile a simple test
  program.

n-west, 2017-02-15:
It appears that you haven't sourced an OE SDK environment file because cmake thinks your compiler is the native one: /usr/bin/gcc

BradleyConn, 2017-02-16:
You are correct. It works now. Thank you!

closed 2017-02-16 by BradleyConn

mikiofuku, 2019-08-01:
Please tell me this solution. change cmake option?

michaelld, 2019-08-01:
@mikiofuku Because this issue is closed: you're better off querying the GR discussion email list, noting this issue in your query.