gnuradio/gnuradio #1217: GSL_FOUND both TRUE and FALSE

anastas, 2017-02-22:
When cmaking "Building for version: v3.7.10.1-237-g81e7af7b / 3.7.11git"
on Fedora 23 (also tested on Fedora 25)  cmake reports that it finds GSL in some modules but it does not find it in some other modules.
Specifically:
--   * gr-dtv
--   * gr-atsc
--   * gr-wavelet
cannot find GSL and are not built.

attached is the output of cmake.
[cmake_out.txt](https://github.com/gnuradio/gnuradio/files/794709/cmake_out.txt)

romeojulietthotel, 2017-02-23:
My experience with these kinds of problems has been that each module seems to use a different technique for gathering dependencies.  In other words the build system is really inconsistent.  I don't know cmake well enough to see a clear way to fix them so I have done work arounds to get the modules to build.  For example  when running cmake I do things like `cmake -DPYTHON_LIBRARY=/drive/dev/python27/lib/libpython2.7.so ../`

One approach to actually fix the problems might be to use ccmake, for example:
`ccmake .`  and then have a look at things.

These problems also plague the cgran modules so I am guessing that the documents/scripts that are used by module creators needs to be fixed.

Many hands make light work, i.e. gnuradio and the modules need more of the users to step up and submit pull requests.  I plan to help where and when I can.

bastibl, 2017-03-26:
If this is still a problem, could you please try adding `set(GSL_FOUND OFF)` at the top of [FindGSL.cmake](https://github.com/gnuradio/gnuradio/blob/master/cmake/Modules/FindGSL.cmake).

I don't know much about cmake, but maybe GSL is not found through pkg_config. Then, in the first run, cmake will look for the library as it is supposed to. In the second, when GSL was already found, the part at the top might assume that it was found through pkg_config and overwrites the already set variables with voids.
```
PKG_CHECK_MODULES(GSL "gsl >= 1.10")
IF(GSL_FOUND)
    set(GSL_LIBRARY_DIRS ${GSL_LIBDIR})
    set(GSL_INCLUDE_DIRS ${GSL_INCLUDEDIR})
ELSE(GSL_FOUND)
```
Then, FIND_PACKAGE_HANDLE_STANDARD_ARGS (at the very bottom) might complain about the void variables and set GSL_FOUND to FALSE. If this is actually the case it should go back and forth between working and not working, but maybe something is cached.

marcusmueller, 2020-10-30:
I think this is fixed through CMake modernization.

closed 2020-10-30 by marcusmueller