gnuradio/gnuradio #1219: QT GUI with id "error"

git-artes, 2017-03-02:
Hi,
I've found this small bug today. When using a QT GUI Widget (tested it with Check Box and Range) which ID is "error", and if a QT GUI Instrumentation is present (tested it with Time Sink and Frequency Sink) one gets a SyntaxError: invalid syntax. The generated python (i.e. top_block.py) code includes (a variation of):

    def set_error(self, error):
        self.error = error        
        self.qtgui_time_sink_x_0.Template self.error: set_time_domain_axis($min, $max)
            min expected 1 arguments, got 0 
        self.qtgui_time_sink_x_0.Template self.error: set_title($which, $title)
            cannot find 'which'      
        self.qtgui_time_sink_x_0.Template self.error: set_color($which, $color)
            cannot find 'which' 
        self._error_callback(self.error)

BTW, I have Ubunut 16.04 and GNU Radio 3.7.10.1. To reproduce the bug I had to simply generate a new flowgraph, include a constant source, one QT GUI Widget (with "error" as its id) and a QT GUI Instrumentation. 
best
Federico

kartikp1995, 2017-03-16:
@git-artes Thanks for notifying the problem.

Update on this issue: It is not necessary to have a QT GUI Widget. Even a `Variable` block with variable name `error` and a QT GUI Instrument will give this error. But QT GUI instrument is necessary. 

Hence, the problem is somewhere in GRC file of QT GUI instruments. Also, while rendering to `top_block.py`, the syntax error in XML decoding occurs at <callback> tag in QT instruments. The reason being: The arguments of callback functions in the XML template can not be decoded by Cheetah. Surprisingly, it is correct. Because the Cheetah variables in the argument of callback functions are not defined in entire template. That's why, BER sink, Number sink, QT GUI Sink, does not throw error. 

So, now this poses 2 questions: 
1. Why does this error occur only when "error" string is defined as a label?
2. Why does this error does not occur in all other cases? There must be the error in those cases, too.

I am not sure how to approach this issue. According to me, this error should be everywhere. Any help would be appreciated.

BrashEndeavours, 2017-08-08:
As @kartikp1995 mentions, I believe when callbacks in the GRC encounter a template error they are written into `callbacks[id]` with `id='error'`.

Normally not an issue as `callbacks['error']` is not accessed....
Except when a variable or other element has an `id='error'`, which causes the erroneous callbacks to be attempted to be rendered [here].(https://github.com/gnuradio/gnuradio/blob/master/grc/core/generator/flow_graph.tmpl#L311-L313)

I propose either:

1. A suitable named token be reserved in the namespace, preventing` 'error'` id collisions; or,
2. Parsing all xml files for unused/misconfigured callbacks, and removing them.

marcusmueller, 2020-10-30:
Don't know whether this problem "survived" the YAML transition, or still exists on maint-3.7

mormj, 2021-03-24:
This one did not seem to survive the YAML transition - I can name anything `error` that I want to now.  If it still exists on maint-3.7 it is unlikely to be fixed, so closing this issue

closed 2021-03-24 by mormj