gnuradio/gnuradio #1222: Update SWIG dependency from 1.3.3 to at least 2.0

marcusmueller, 2017-03-09:
Revisiting https://bugzilla.redhat.com/show_bug.cgi?id=1143914 . 

Gist of it:

runtime/thread/thread.h uses `int` where it should use `size_t`. We didn't adjust that (back then) because we weren't sure it'd break older SWIGs, though it would've been the _right thing to do_<sup>TM</sup>. It breaks build on a few "exotic" archs (s390), but not having that patch around just because it _might_ break something on systems *older than RedHat 5, Ubuntu 12.04 or Fedora 14* makes little to no sense; noone uses pre-2.0.4 SWIG anymore. Patch is really minimal, attached

[gnuradio-3.7.10-size_t.txt](https://github.com/gnuradio/gnuradio/files/831620/gnuradio-3.7.10-size_t.txt)

If we upstream that, GR builds unpatched on all architectures supported by RedHat, CentOS, Fedora.

If we bump the dependency, I'll make a PR for said patch against master and next.

marcusmueller, 2017-03-09:
For perspective: SWIG versions offered by old distros; in **bold** the pre-2.0 SWIGs.

| Distro | EOL | Version of SWIG |
|------------------------|--------------|-------------------------|
| CentOS 5/RHEL 5 |2013-01-08| **1.3.29-2.el5** |
| Debian 6 (squeeze) | 2014-05-31 (full) / 2016-02-29 (LTS) | **1.3.40-3** |
| Ubuntu 10.04 LTS| 2013-05-09 (Desktop) / 2015-04-30 (Server) | **1.3.40-2ubunut1**|
| CentOS 6/RHEL 6 | 2016-05-10| **1.3.40-6.el6** |
| **EOL LINE**| <hr> | <hr>|
| Ubuntu 12.04 LTS| 2017-04-XX | 2.0.4+really2.0.4-4ubuntu2 (gotta love Canonical's versioning)|
| **EOL @ GR 3.8 Release LINE** | <hr> | <hr>|
| Debian 7 (wheezy/*oldstable*) | 2016-04-26 (full) / 2018-05-XX (LTS)) | 2.0.7 |
| Ubuntu 14.04 LTS | 2019-04 | 2.0.11-1ubuntu2|
| CentOS 7/RHEL 7 | 2019-Q4 | 2.0.10-5.el7 |
| MacPorts time-of-writing | <hr> | 3.0.12 |

jmcorgan, 2017-03-09:
Yeah, it looks like we can safely bump to 2.0.4 for 3.7.x.

mbr0wn, 2017-03-26:
On master, it's 2.0.4 now, and we have other PRs lined up for this. This PR is now redundant.

mbr0wn, 2017-03-26:
...this issue, I mean.

closed 2017-03-26 by mbr0wn