gnuradio/gnuradio #1227: can't use boost::(random::)uniform_int_distribution: requires boost version >> minimal_boost_version

marcusmueller, 2017-03-10:
44ef5e1c240b7d5aeac7ca6537c4b07f3cf606c5  added new analog random generators, functionally based on `boost::random::uniform_int_distribution`.

Problem is that uniform_int_distribution only became available in a version `v`, 1.45.5 < v <= 1.48 . Since our current Boost version requirement is 1.35, and not overly likely to become 1.48 soon, we simply can't do that and must revert the aformentioned patch, lest we find a way around `uniform_int_distribution`.

There's a class `uniform_int` that is in 1.41¹, which is deprecated in modern boosts. Maybe it does the same. Haven't looked at it, because I'm not sure what the code does, calling @jdemel on that 👸 !

<hr>
¹ by the way, let's soon as possible bump the Boost requirement to at least that

jdemel, 2017-03-10:
This is very unfortunate.

First off, is this fix necessary for GR3.8?
IMHO, only GR3.7 needs to be fixed. RHEL6 enters 'Production Phase 3' by May '17.
[https://en.wikipedia.org/wiki/Red_Hat_Enterprise_Linux]

It seems like the 'random' class needs to be fixed. More precisely the 'ran_int' method.  
Besides boost::mt19937 should be usable. Also, this implies that the newly introduced block 'random_uniform_source' does not need to be fixed.

The used boost::random functions are:
boost::uniform_real<...>
boost::random::uniform_int_distribution<>(min, max)
boost::variate_generator<...>

Is 'boost::random::uniform_int_distribution' the only function that causes trouble? Or does the use of the whole boost::random system imply compatibility issues? @stwunsch introduced MT to gr::random. He might comment on it as well.

There might be a way to use random numbers from boost::mt19937 directly. But that implies min/max values need to be checked in 'ran_int'. Bound checking is delegated to boost which is the whole point of using this function here.

marcusmueller, 2017-03-10:
>  RHEL6 enters 'Production Phase 3' by May '17. 

If only that was our only problem, #1228 

> It seems like the 'random' class needs to be fixed. More precisely the 'ran_int' method.

yeah, and I think some of the things in gr-analog, but not sure, so that's why I'm asking you as an expert :)

>  boost::random::uniform_int_distribution<>(min, max)

That's the one non-present in Boost 1.41, where I stumbled across this.

> Is 'boost::random::uniform_int_distribution' the only function that causes trouble?

Can't say that for sure until we've fixed this single problem. I'm not a C preprocessor nor a C++ compiler :D 

> @stwunsch introduced MT to gr::random. He might comment on it as well.

Nope, boost::random was there before. In fact, I remember fixing the issue that we used boost::random::mt19937 instead of the "old" boost::mt19937 namespacing, which was an ugly thing to revert to, but fixed things for pre-1.46 boosts.

>There might be a way to use random numbers from boost::mt19937 directly. But that implies min/max values need to be checked in 'ran_int'. Bound checking is delegated to boost which is the whole point of using this function here.

Hm, don't know how we can fix that.

jdemel, 2017-03-10:
We should bump the minimum boost version. The newer the better.
GR3.8 should at least require something like ~boost.v1.54

Besides, there is
'http://www.boost.org/doc/libs/1_40_0/boost/random/uniform_int.hpp'

This should fix our problems. At least the header states that it dates back to 2001.

marcusmueller, 2017-03-10:
> Besides, there is 'http://www.boost.org/doc/libs/1_40_0/boost/random/uniform_int.hpp'
> This should fix our problems. At least the header states that it dates back to 2001.

No; read the header. No `uniform_int_distribution` in there, only `uniform_int`, of which I don't know if it can work as a replacement for `uniform_int_distribution` (which is why I pinged you from the original issue in the first place!).

jdemel, 2017-03-10:
It seems like` uniform_int` may be a replacement for `uniform_int_distribution`. That's why I suggest using `uniform_int`. This needs to be tested.

marcusmueller, 2017-03-10:
> It seems likeuniform_int may be a replacement for uniform_int_distribution. That's why I suggest using uniform_int. This needs to be tested.

Seriously, that's literally exactly what I said in the third paragraph of the issue report. That's why I brought you here :)

jmcorgan, 2017-03-10:
FYI, I think we've settled on Boost >=1.53 for 3.8.

jdemel, 2017-03-13:
That's good news, I guess. Is there an overview page for GR3.8 mimimum requirements?

jmcorgan, 2017-03-16:
Not yet, but we're working it out in this doc:

https://docs.google.com/spreadsheets/d/1N5zlMyxE8qfmaL6HTI5oFFVU47-JC3cPJgpvoiWTCg0/edit?usp=sharing

mbr0wn, 2017-03-26:
I lost track of what the problem is. If we bump Boost to 1.53, is this a none-issue then? Because if so, we can close this.

jmcorgan, 2017-03-26:
We merged #1123 anyway, so it's no longer an issue.

PR #1231 (2017-03-21): replaced uniform_int_distribution with uniform_int
This should fix #1227
closed 2017-03-26 by jmcorgan