gnuradio/gnuradio #1228: Master: Update Boost Dependency to 1.48

marcusmueller, 2017-03-10:
For perspective: Boost versions offered by old distros

| Distro | EOL | Version of SWIG |
|------------------------|--------------|-------------------------|
| [Debian 6 (squeeze)](https://archive.debian.net/squeeze) | 2014-05-31 (full) / 2016-02-29 (LTS) |  |
| [Ubuntu 10.04 LTS](http://old-releases.ubuntu.com/ubuntu/dists/lucid/main/binary-amd64/Packages.bz2)| 2013-05-09 (Desktop) / 2015-04-30 (Server) | 1.40–mix-Ubuntu-eff-up |
| CentOS 6/RHEL 6 | 2016-05-10|  1.41 |
| **EOL LINE**| <hr> | <hr>|
| Ubuntu 12.04 LTS| 2017-04-XX | 1.48-Ubuntu-versioning-eff-up |
| **EOL @ GR 3.8 Release LINE** | <hr> | <hr>|
| [Debian 7 (wheezy/*oldstable*)](https://packages.debian.org) | 2016-04-26 (full) / 2018-05-XX (LTS)) |  1.49.0.1|
| Ubuntu 14.04 LTS | 2019-04-XX | 1.54.0.1 |
| CentOS 7/RHEL 7 | 2019-Q4 | 1.53.0 |

marcusmueller, 2017-03-10:
### Personal Assessment

| Branch | my rec. minimal Boost|
|---|---|
| master/3.7.10.X | 1.48-Ubuntu-whatevs |
| next | 1.53.0|

I'm positive we can update our boost dependency right now to at least 1.41 without affecting a build, and should probably push it to 1.53 for next (I don't think it'll be hard to explain that GR3.8 isn't supported on debian oldstable). For the v3.7.X family, we can push it to 1.48, even if it happens before EOL Ubuntu 12.04.

jmcorgan, 2017-03-10:
I'll have to recheck, but I think Boost 1.48 had a serious threading issue and we had to blacklist that version. If that's the case then bumping this to 1.49 for master looks good.

mbr0wn, 2017-03-19:
GrBoost.cmake:
```{.cmake}
# For any unsuitable Boost version, add the version number below in                                                             
# the following format: XXYYZZ                                                                                                  
# Where:                                                                                                                        
#     XX is the major version ('10' for version 1)                                                                              
#     YY is the minor version number ('46' for 1.46)                                                                            
#     ZZ is the patcher version number (typically just '00')                                                                    
set(Boost_NOGO_VERSIONS                                                                                                         
  104600 104601 104700 105200                                                                                                   
  )                                                                                                                             
```

Looks like 1.48 is not on the list.

jmcorgan, 2017-03-19:
Got it, must have been thinking of the 1.46-1.47 days.  In any case, we can update 3.7.x to 1.53 when Ubuntu 12.04 goes out of support in a couple weeks.

mbr0wn, 2017-03-26:
If we merge #1252, we get this. I'll close this and move the action to the PR.

closed 2017-03-26 by mbr0wn