gnuradio/gnuradio #1229: gr-uhd: remove non-stream-api case?

marcusmueller, 2017-03-10:
These `#ifdefs` have been bugging me for quite a while, and I'm not even sure anyone does test builds let alone functional tests with the old API. 

As far as I can see, we should clean these out for 3.8, but unlike @mbr0wn I might be lacking the comprehensive understanding of why they're in there.

In the same step, we should probably require a minimum UHD version of 3.7; I'm not sure we're doing anyone a favour by allowing builds with 3.5.2.

jmcorgan, 2017-03-10:
I know @mbr0wn is already on this--Martin, can you summarize the plan?

mbr0wn, 2017-03-11:
3.7 is pretty new in terms of what distros have. `next` already removed a lot of `#ifdefs`, but I really don't think we can go all the way up to 3.7 (as much as I would like to).

mbr0wn, 2017-03-11:
...the non-streamer-API case is gone, though (it's already merged). So I think we can close this.

closed 2017-03-11 by mbr0wn