gnuradio/gnuradio #1230: Move source blocks from gr-analog to gr-blocks

jdemel, 2017-03-10:
The blocks:
const_source_x
fastnoise_source_x
noise_source_x
random_source_x
random_uniform_source_x
sig_source_x

should be moved to gr-blocks.

Reason: They are essential to most signal processing flowgraphs. And they are not strictly 'analog' blocks.
Furthermore, the random sources should be renamed. Their names should reflect the distribution.

jmcorgan, 2017-03-10:
If anything, I'd be in favor of breaking gr-blocks up into smaller groups (we already have to split it into 10 separate SWIG builds for memory reasons).

That aside, renaming these to have more descriptive/intuitive names would be useful--what is your suggestion?

marcusmueller, 2017-03-10:
Personally, I'd like to have **one** `make` factory for different types of noise, not one make per impl of the gengen templates for each type, and then branching in the `work`. The question is how to make that visible in doxygen; but I'm sure we can come up with something.

So, my idea would actually be having a single public class ("noise_source"), with the following factory function

    foo_sptr make( enum {UNIFORM, GAUSSIAN, LAPLACIAN, WHATEVS} dist,
         enum {FLOAT, INT, GR_COMPLEX, BYTE} type,
         float paramA, float paramB,
         size_t precompute_buffer_size=0)

For most of these things, we, by now, have heavily specialized impls – see @stwunsch's boot::random improvements, so I think it'd be a pretty good idea to get rid of the gengen.

Obviously, we'd need message handlers for run-time parameter adjustments (can't know the specific methods if you can't know which kind of source you're handling) – but these would then, at least, be coherent.

jdemel, 2017-03-13:
Given that gr-blocks is too big. We might introduce gr-source. Move all source blocks, that provide some general functionality into this module.

Naming:
constant_source
fast_complex_gaussian_source (or rayleigh)
complex_gaussian_source
fast_uniform_random_source
uniform_random_source
signal_source (or a better name)

marcusmueller, 2017-03-13:
I like the "put functionally similar things into the same module" approach. Naming would be gr-siggen to avoid confusion with the general "this block is a source" concept.

@jmcorgan gut feeling: since that is really a big issue (it doesn't stop at gr-analog/gr-blocks; what about gr-digital, gr-filter, gr-wavelet, gr-fec and gr-trellis; should we have a gr-resampler?) could you make a "repartition modules" issue. I think we can extract all the blocks we have simply by dumbly parsing the lib/ directories. I have ideas in my head (I know, terrible, it never stops) about how we can have a somewhat graphical, yet easily verifiable and code-translatable, way to cooperatively deal with where to put which block.

mhostetter, 2017-03-28:
I also think breaking up `gr-blocks` is a good idea. I also like @marcusmueller's idea of "putting functionally similar blocks in the same module".

What about something like:

* `gr-controlport` for Control Port blocks
* `gr-conversion` for Type Conversion blocks
* `gr-debug` for Debug Tools blocks
* `gr-file` for File Operator blocks
* `gr-math` or `gr-arithmetic` for Math Operators blocks
* `gr-message` for Message Tools blocks
* `gr-stream` for Stream Operators blocks

jmcorgan, 2017-04-02:
I don't like proliferating top-level components, as they impact a lot of things throughout the tree (inter-component dependencies, cmake, documentation, import lines, etc.)  So, while I'm not fond of expanding gr-blocks, I think the simplest thing would be to follow @jdemel original idea (move some from gr-analog to gr-blocks).

mbr0wn, 2017-07-14:
Because so many people use them, this'll break a lot of people's applications. I agree with the sentiment, but is it worth it?

marcusmueller, 2017-07-14:
Well, if not now, then when? Personally, I consider 3.8 a chance to break API where we can in order to make things better. Plus, if breakage sounds scary, at least for Pythonland, we can easily alias (and warn), so that only C++ hier blocks actually break

jmcorgan, 2017-07-14:
@marcusmueller Can write short but concrete description of the change you're proposing? I agree now (3.8) is the time to break things, but let's see what you actually have in mind.

marcusmueller, 2017-07-14:
@jmcorgan I can, but I'm literally on a hot train without a table. A phone is not the perfect tool for conciseness, but I'll put that on my agenda for the weekend.

jmcorgan, 2017-07-14:
No worries.

marcusmueller, 2017-07-25:
@jmcorgan coming up with an approach:

Restructuring of GNU Radio in-tree modules
------------------------------------------

Goal
====

* Make location of blocks more intuitive to the user
* Make GNU Radio faster to build (smaller linker and SWIG objects)
* Make Maintenance easier by having assignable subsystems

Approach
========

* Move semantically similar blocks into logically named modules
* Avoid cross-dependencies between modules, aside from unittests and very common blocks
* Where breakage of end-user applications is to be expected, add a temporary alias to the individual Python modules, so that one can import the block from its old position, but gets a deprecation warning
* Don't be afraid of having more separate modules. 

Proposed Structure
==================

diffs for illustration; generated by manually cut'n'pasting text generated by 

    >>> from gnuradio import MODULENAME
    >>> l = [ k[:-5]+"\n"  for k in  MODULENAME.__dict__.keys() if k.endswith("_make")]
    >>> f = open("MODULENAME.txt", "w")
    >>> f.writelines(l)
    >>> f.close()

* gr-blocks *should* shrink
* gr-math should get all the zero-dependencies (aside from VOLK/Boost) basic math blocks
* gr-fileio gets... file io. I'd like to see more file sinks/sources upstreamed to that, and those being dynamically enabled on a per-block basis. E.g. a WAVPACK source/sink (relatively good compression for IQ data), MIT Haystack… This would however be a prime candidate for aggressive maintenance, ie. automated tests with test files. 
* gr-streamops: things like stream_to_vector, head, you get the idea.  zero-deps
* gr-logical kind of on the border to gr-math, should maybe get the packed/unpacked? zero-deps
* gr-filter gets the moving_average, which for some reason is in gr-blocks

Not included in this draft (yet):

* gr-sigsources: noise sources, signal_source

**gr-blocks**
__________

  --- blocks.txt	2017-07-25 16:55:17.213755879 +0200
  +++ blocks_new.txt	2017-07-25 17:08:47.707595894 +0200
  @@ -1,232 +1,34 @@
  -abs_…
  -add_const_bb …
  -add_…
  -and_bb 
  -and_const_…
  -and_…
  -annotator_1to1 …
  -argmax_…
  bin_statistics_f
  -burst_tagger
  -char_to_…
  check_lfsr_32k_s
  -complex_to_…
  -conjugate_cc
  -copy
  ctrlport_probe2_…
  -deinterleave
  -delay
  -divide_…
  -endian_swap
  -exponentiate_const_cci
  -file_descriptor_sink
[... 297 lines cut ...]
___________

  --- fileio.txt	2017-07-25 17:00:54.823520916 +0200
  +++ fileio_new.txt	2017-07-25 17:03:14.550906905 +0200
  @@ -0,0 +1,9 @@
  +file_descriptor_sink
  +file_descriptor_source
  +file_meta_sink
  +file_meta_source
  +file_sink
  +file_source
  +tagged_file_sink
  +wavfile_sink
  +wavfile_source

**gr-filter**
_______________

  --- filter.txt	2017-07-25 17:10:47.873647748 +0200
  +++ filter_new.txt	2017-07-25 17:10:39.083497654 +0200
  @@ -33,6 +33,10 @@
  +moving_average_cc
  +moving_average_ff
  +moving_average_ii
  +moving_average_ss

marcusmueller, 2023-10-24:
As we pull up a block library for GR4, we get the unique chance to sensibly organize them. Let's do that.