gnuradio/gnuradio #1236: Impossibility to disable DC filtering for USRP source

pelijah, 2017-03-11:
There is an error in the script _gr-uhd/grc/gen_uhd_usrp_blocks.py_. To disable DC filtering user needs to enter _False_ in the field  _$dc_offs_enbXX_. In this case _set_auto_dc_offset_ will not be called because condition "if False" is always false.
```
	\#if \$dc_offs_enb$(n)()
self.\$(id).set_auto_dc_offset(\$dc_offs_enb$(n), $n)
	\#end if
```
It needs to be changed on something like this
	`\#if \$dc_offs_enb$(n)() != ""`

mbr0wn, 2017-03-19:
Yep, that's legit. @pelijah do you want to submit a PR for this?

pelijah, 2017-03-20:
@mbr0wn No

marcusmueller, 2021-01-16:
closing this two years after a fix was merged (if it wasn't fixed by that fix, the bug reporter would probably have spoken out?)

closed 2021-01-16 by marcusmueller