gnuradio/gnuradio #1238: [volk/apps/volk_profile] Error 1

peterwillcn, 2017-03-12:
volk_32f_x2_subtract_32f_a_orc_impl.c:(.text+0x5b9): undefined reference to `orc_program_free'
collect2: error: ld returned 1 exit status
volk/apps/CMakeFiles/volk_profile.dir/build.make:125: recipe for target 'volk/apps/volk_profile' failed
make[2]: *** [volk/apps/volk_profile] Error 1
CMakeFiles/Makefile2:348: recipe for target 'volk/apps/CMakeFiles/volk_profile.dir/all' failed
make[1]: *** [volk/apps/CMakeFiles/volk_profile.dir/all] Error 2
[ 34%] Building CXX object gr-qtgui/lib/CMakeFiles/gnuradio-qtgui_static.dir/edit_box_msg_impl.cc.o
[ 34%] Building CXX object gr-blocks/lib/CMakeFiles/gnuradio-blocks_static.dir/integrate_ff_impl.cc.o

n-west, 2017-03-12:
You've provided no information to help resolve the error. It appears that there's a failure to link against ORC. Do you have ORC installed? Where? What environment are you building VOLK in?

marcusmueller, 2017-03-13:
Attaching the complete `CMake` and `make` outputs as text files might be appropriate, along with version information.

peterwillcn, 2017-03-20:
use -DENABLE_STATIC_LIBS=True
see https://github.com/gnuradio/volk/issues/111

closed 2017-03-20 by peterwillcn