gnuradio/gnuradio #1239: block: Add "message" type to null source/sink

mhostetter, 2017-03-13:
Sometimes when debugging message passing blocks, it's useful to not trigger the downstream processing.  The Message Debug block works but can clutter stdout with print statements.  I found the ZMQ Message PUB block works, but is kind of a hack.

jmcorgan, 2017-03-13:
To clarify, then you'd like a null sink for messages, which would be a new block, that would then get selected in GRC by using the "message" type in the configuration dialog? (This is how it works for the other types for the streaming ports.)

mhostetter, 2017-03-14:
Yes.  I realize I wasn't very clear.  I'm suggesting adding "Message" as an Input or Output Type to the already existing Null Sink and Source blocks.  Currently it only supports the streaming ports.

[image]

I just wanted to submit this thought before I forgot it.  I might have time to take a whack at implementing this, unless someone else jumps on it.

jmcorgan, 2017-03-14:
Got it.

The actual C++ null sink block doesn't have a type; it takes an itemsize parameter, and only has a streaming port.  The GRC wrapper creates the above options but just turns them into an itemsize value to pass into the Python interface to the block.

To implement the above, you'll need to write a new C++ block that has a message port instead of a streaming port, then in the message handler, does nothing.  You'll also need to modify the GRX XML file such that it offers the message type, and in the this case, selects this block instead of the other one for the <make></make> entry.

It's a straightforward effort and would give you a chance to both learn more about a GNU Radio internals and contribute to the project.

mhostetter, 2017-03-14:
Great!  Thanks for the direction.  I'll start working on it.

Keep an eye out for a pull request!  :smile:

n-west, 2017-03-14:
The expected behavior of a null message source is fairly obvious, but what are you expecting from a null message source?

mhostetter, 2017-03-14:
Well maybe I am misunderstanding something, but here's what I was thinking.

I created a custom source block with an optional message output port.  It's `timed_source.py` in my (unfinished) OOT module [gr-sqlite](https://github.com/mhostetter/gr-sqlite).  It was designed to format and output PDUs from a SQLite database in realtime, timed by a UNIX timestamp in the database.  As such, it has no input, just the PDU output.  (I'm actually also having trouble with that -- creating child threads in a Python block).

What I believe I discovered was if the PDU output was not connected to anything (because it was optional), the block wouldn't run.  Maybe GRC optimized the block out because it had no port connections?  So I couldn't test the block.  To get the block to run I connected the PDU output to a Message Debug input, but then the stdout was overwhelmed with PDU print statements.

The goal I had in mind was for GRC to think the message source had a sink (the null sink) so it would connect it in the top block, but the sink would drop the message on the floor.

Do I have the GRC functionality right?

Thanks for the help.

mhostetter, 2017-03-14:
I just realized you asked about the Null Source and I answered about the Null Sink.  As for the Null Source, I actually don't know.  I might not have any desire for that...

bastibl, 2017-03-26:
I think also the sink version is not needed. You just make the output optional as you already mentioned. GNU Radio won't optimize out any blocks.
AFAIS, the problem is your block. Your `start()` never returns as it directly calls `run()`. Note the blocks main thread belongs to the GNU Radio scheduler. You should not try to overwrite and manually `run()` it . You might want something similar to GNU Radio's Message Strobe. It creates a new thread in `start()`. This thread is not managed by the scheduler and is used to inject PDUs from time to time.

mbr0wn, 2017-03-27:
Agree with @bastibl. We have a pub/sub model for the messages, so if there's no sub, that doesn't really matter from a GNU Radio perspective.

jmcorgan, 2017-03-27:
Ok, let's close this out then.

closed 2017-03-27 by jmcorgan

mhostetter, 2017-03-27:
Ok, thanks for the feedback.